OpenAMP/open-amp #8: Follow openamp_gs.pdf and meet build error.

kinreven, 2015-08-28:
openamp_gs.pdf:
5. Build PetaLinux
a. Change into the <master_root>
          $ cd <master_root>
b. Execute the following command:
          $ petalinux-build

kinreven@ubuntu:~/openamp/xilinx-zynq-linux$ petalinux-build
INFO: Checking component...
INFO: Generating make files and build linux
INFO: Generating make files for the subcomponents of linux
INFO: Building linux
[INFO ] pre-build linux/rootfs/rpmsg_echo_test_kern_app
[INFO ] pre-build linux/rootfs/rpmsg_mat_mul_kern_app
[INFO ] pre-build linux/rootfs/rpmsg_proxy_dev_driver
[INFO ] pre-build linux/rootfs/rpmsg_user_dev_driver
[INFO ] pre-build linux/rootfs/zynq_remoteproc_driver
[INFO ] pre-build linux/rootfs/echo_test
[INFO ] pre-build linux/rootfs/echo_test_baremetal_fw
[INFO ] pre-build linux/rootfs/fwupgrade
[INFO ] pre-build linux/rootfs/mat_mul_baremetal_fw
[INFO ] pre-build linux/rootfs/mat_mul_demo
[INFO ] pre-build linux/rootfs/peekpoke
[INFO ] pre-build linux/rootfs/proxy_app
[INFO ] pre-build linux/rootfs/rpc_demo_baremetal_fw
[INFO ] pre-build linux/rootfs/uWeb
[INFO ] build system.dtb
[INFO ] build linux/kernel
[INFO ] update linux/u-boot source
[INFO ] generate linux/u-boot configuration files
[INFO ] build linux/u-boot
[INFO ] Setting up stage config
[INFO ] Setting up rootfs config
[INFO ] Updating for armv7a-vfp-neon
[INFO ] Updating package manager
[INFO ] Expanding stagefs
[INFO ] build kernel in-tree modules
[INFO ] modules linux/kernel
[INFO ] build linux/rootfs/rpmsg_echo_test_kern_app
[INFO ] modules linux/kernel
[INFO ] build linux/rootfs/rpmsg_mat_mul_kern_app
[INFO ] modules linux/kernel
[INFO ] build linux/rootfs/rpmsg_proxy_dev_driver
[INFO ] modules linux/kernel
[INFO ] build linux/rootfs/rpmsg_user_dev_driver
[INFO ] modules linux/kernel
[INFO ] build linux/rootfs/zynq_remoteproc_driver
[INFO ] modules linux/kernel
[ERROR] make[6]: *** [/home/kinreven/openamp/xilinx-zynq-linux/build/linux/rootfs/modules/zynq_remoteproc_driver/zynq_remoteproc_driver.o] Error 1
[ERROR] make[5]: *** [_module_/home/kinreven/openamp/xilinx-zynq-linux/build/linux/rootfs/modules/zynq_remoteproc_driver] Error 2
[ERROR] make[4]: *** [sub-make] Error 2
[ERROR] make[3]: *** [modules_only] Error 2
[ERROR] make[2]: *** [modules] Error 2
[ERROR] make[1]: *** [sub_build_component_/none/modules/multi/zynq_remoteproc_driver] Error 2
ERROR: Failed to build linux

wjliang, 2015-08-28:
which petalinux version do you use? if you are using petalinux 2014.4 or above, and if you are using Linux as master, please do not using the zynq_remoteproc driver from this openamp git, but use the zynq_remoteproc directly from the kernel. You can run "petalinux-config -c kernel" from petalinux, go to remoteproc submenu, and select the driver. The Zynq remoteproc in this repo hasn't been updated for the newer kernel. Prefer to remove it from here, and update the document to use the one directly from Xilinx kernel

kinreven, 2015-08-28:
@wjliang  i am using petalinux 2013.10 and using linux master. i believe that i found the root cause of the build error, as blow. but i have no ider why my kernel miss this header file(inux/irqchip/arm-gic.h)?

kinreven@ubuntu:~/openamp/zynq-amp-linux/build/linux/rootfs/modules/zynq_remoteproc_driver$ make
[INFO ] build linux/rootfs/zynq_remoteproc_driver
make INSTANCE=linux/kernel -C /home/kinreven/openamp/zynq-amp-linux/build/linux/kernel M=/home/kinreven/openamp/zynq-amp-linux/build/linux/rootfs/modules/zynq_remoteproc_driver modules_only; \

make[1]: Entering directory `/home/kinreven/openamp/zynq-amp-linux/build/linux/kernel'
rm -rf /home/kinreven/openamp/zynq-amp-linux/build/linux/kernel/boot /home/kinreven/openamp/zynq-amp-linux/build/linux/kernel/usr
ln -s "/home/kinreven/openamp/zynq-amp-linux/build/linux/kernel/xlnx-3.8/arch/arm/boot" "/home/kinreven/openamp/zynq-amp-linux/build/linux/kernel/boot"
ln -s "/home/kinreven/openamp/zynq-amp-linux/build/linux/kernel/xlnx-3.8/usr" "/home/kinreven/openamp/zynq-amp-linux/build/linux/kernel/usr"
[INFO ] modules linux/kernel
make ARCH=arm CROSS_COMPILE=arm-xilinx-linux-gnueabi- O=/home/kinreven/openamp/zynq-amp-linux/build/linux/kernel/xlnx-3.8 KCONFIG_CONFIG="/home/kinreven/openamp/zynq-amp-linux/build/linux/kernel/xlnx-3.8"/.config -j4 -C /opt/petalinux-v2013.10-final/components/linux-kernel/xlnx-3.8 modules
make[2]: Entering directory `/opt/petalinux-v2013.10-final/components/linux-kernel/xlnx-3.8'
  CC [M]  /home/kinreven/openamp/zynq-amp-linux/build/linux/rootfs/modules/zynq_remoteproc_driver/zynq_remoteproc_driver.o
/home/kinreven/openamp/zynq-amp-linux/build/linux/rootfs/modules/zynq_remoteproc_driver/zynq_remoteproc_driver.c:31:35: fatal error: linux/irqchip/arm-gic.h: No such file or directory
compilation terminated.
make[4]: *** [/home/kinreven/openamp/zynq-amp-linux/build/linux/rootfs/modules/zynq_remoteproc_driver/zynq_remoteproc_driver.o] Error 1
make[3]: *** [_module_/home/kinreven/openamp/zynq-amp-linux/build/linux/rootfs/modules/zynq_remoteproc_driver] Error 2
make[2]: *** [sub-make] Error 2
make[2]: Leaving directory `/opt/petalinux-v2013.10-final/components/linux-kernel/xlnx-3.8'
make[1]: *** [modules_only] Error 2
make[1]: Leaving directory `/home/kinreven/openamp/zynq-amp-linux/build/linux/kernel'
make: *** [modules] Error 2

kinreven, 2015-08-29:
@wjliang , i found the linux kernel of petalinux 2013.10 is 3.8. In this kernel version, we should include "asm/hardware/gic.h", and not "linux/irqchip/arm-gic. h". so i guess this remoteproc driver is for linux 3.10 or above. as what you commented, the open-AMP framework could work with petalinux 2014.4 or above. and could you update the getting started documents for betalinux 2015.2? 

BTW, if you add a section how to run this AMP system on qemu, i believe that more and more people can  get started with the project.

wjliang, 2015-08-30:
I just checked the driver inside the open-amp, the last commit shows it works with kernel 3.17, which is matches the one with petalinux v2014.4. If you want to use 3.8 kernel, you can use this HEAD: 211f8add65c15af48543da8423d541fc3ac62006

Sure, we will add AMP system on QEMU later. Just last time when I tested  it, it runs very very slow in QEMU,

kinreven, 2015-08-31:
@wjliang,  i have work well with petalinux 2013.10 by QEMU.

Thank you very much.

closed 2015-08-31 by kinreven

joshkern, 2015-10-08:
@wjliang Is there a version of OpenAMP that works with Peatlilnux 2014.2 (kernel version 3.14)? I tried using both the current version and the commit you posted above but both are not working for me. If I use the 3.8 version, I get the same error kinreven got during petalinux-build. If I use the current version, I get this error during modprobe:

root@Avnet-Digilent-ZedBoard-2014_2:~# modprobe zynq_remoteproc_driver firmware=
"/lib/firmware/zc702evk/baremetal/matrix_multiply/firmware"
CPU1: shutdown

=================================
[ INFO: inconsistent lock state ]
3.14.2-xilinx #14 Tainted: G           O
---------------------------------
inconsistent {IN-HARDIRQ-W} -> {HARDIRQ-ON-W} usage.
modprobe/783 [HC0[0]:SC0[0]:HE1:SE1] takes:
 (irq_controller_lock){?.-...}, at: [<8021a374>] gic_set_affinity+0x60/0xac
{IN-HARDIRQ-W} state was registered at:
  [<8005f6f4>] __lock_acquire+0x75c/0x199c
  [<80061030>] lock_acquire+0x68/0x7c
  [<80495eb0>] _raw_spin_lock_irqsave+0x50/0x64
  [<8021a974>] gic_raise_softirq+0x14/0x80
  [<80014560>] smp_send_reschedule+0x30/0x3c
  [<80049ee8>] check_preempt_curr+0x54/0x80
  [<80049f28>] ttwu_do_wakeup+0x14/0xb0
  [<8004d7bc>] try_to_wake_up+0x244/0x28c
  [<8005aa7c>] autoremove_wake_function+0xc/0x34
  [<8005a4ac>] __wake_up_common+0x4c/0x78
  [<8005a6cc>] __wake_up+0x34/0x48
  [<8008feb4>] __irq_work_run+0x90/0xc8
  [<800144ec>] handle_IPI+0x68/0xac
  [<8000854c>] gic_handle_irq+0x58/0x60
  [<80012444>] __irq_svc+0x44/0x78
  [<80496058>] _raw_spin_unlock_irqrestore+0x40/0x74
  [<80072720>] rcu_process_callbacks+0x190/0x548
  [<80026cc0>] __do_softirq+0x120/0x278
  [<80027104>] irq_exit+0x84/0xf4
  [<8000efa8>] handle_IRQ+0x6c/0x90
  [<80008530>] gic_handle_irq+0x3c/0x60
  [<80012444>] __irq_svc+0x44/0x78
  [<800678a4>] vprintk_emit+0x480/0x504
  [<8048f15c>] printk+0x2c/0x3c
  [<806932d8>] devtmpfs_init+0xb0/0xdc
  [<806931b0>] driver_init+0x8/0x28
  [<8067cba8>] kernel_init_freeable+0x98/0x1d0
  [<8048bcdc>] kernel_init+0x8/0xe4
  [<8000e728>] ret_from_fork+0x14/0x2c
irq event stamp: 9995
hardirqs last  enabled at (9995): [<804960b0>] _raw_spin_unlock_irq+0x24/0x5c
hardirqs last disabled at (9994): [<80495edc>] _raw_spin_lock_irq+0x18/0x60
softirqs last  enabled at (9706): [<80026d90>] __do_softirq+0x1f0/0x278
softirqs last disabled at (9697): [<80027104>] irq_exit+0x84/0xf4

other info that might help us debug this:
 Possible unsafe locking scenario:

       CPU0
       ----
  lock(irq_controller_lock);
  <Interrupt>
    lock(irq_controller_lock);

 *** DEADLOCK ***

2 locks held by modprobe/783:
 #0:  (&__lockdep_no_validate__){......}, at: [<80271be8>] __driver_attach+0x34/0x94
 #1:  (&__lockdep_no_validate__){......}, at: [<80271c0c>] __driver_attach+0x58/0x94

stack backtrace:
CPU: 0 PID: 783 Comm: modprobe Tainted: G           O 3.14.2-xilinx #14
[<80015658>] (unwind_backtrace) from [<800119b4>] (show_stack+0x10/0x14)
[<800119b4>] (show_stack) from [<80490688>] (dump_stack+0x8c/0xd0)
[<80490688>] (dump_stack) from [<8048eda0>] (print_usage_bug+0x25c/0x2d0)
[<8048eda0>] (print_usage_bug) from [<8005e6bc>] (mark_lock+0x3d0/0x6b8)
[<8005e6bc>] (mark_lock) from [<8005f760>] (__lock_acquire+0x7c8/0x199c)
[<8005f760>] (__lock_acquire) from [<80061030>] (lock_acquire+0x68/0x7c)
[<80061030>] (lock_acquire) from [<80495d68>] (_raw_spin_lock+0x40/0x50)
[<80495d68>] (_raw_spin_lock) from [<8021a374>] (gic_set_affinity+0x60/0xac)
[<8021a374>] (gic_set_affinity) from [<8021a3fc>] (gic_set_cpu+0x3c/0x44)
[<8021a3fc>] (gic_set_cpu) from [<7f093470>] (zynq_remoteproc_probe+0x1c8/0x3ec [zynq_remoteproc_driver])
[<7f093470>] (zynq_remoteproc_probe [zynq_remoteproc_driver]) from [<80273114>] (platform_drv_probe+0x18/0x48)
[<80273114>] (platform_drv_probe) from [<80271958>] (driver_probe_device+0x11c/0x334)
[<80271958>] (driver_probe_device) from [<80271c24>] (__driver_attach+0x70/0x94)
[<80271c24>] (__driver_attach) from [<8026ff14>] (bus_for_each_dev+0x70/0x84)
[<8026ff14>] (bus_for_each_dev) from [<80270ffc>] (bus_add_driver+0xfc/0x1f4)
[<80270ffc>] (bus_add_driver) from [<8027223c>] (driver_register+0x9c/0xe0)
[<8027223c>] (driver_register) from [<800088b8>] (do_one_initcall+0xb8/0x160)
[<800088b8>] (do_one_initcall) from [<80086070>] (load_module+0x1744/0x1d48)
[<80086070>] (load_module) from [<80086770>] (SyS_init_module+0xfc/0x10c)
[<80086770>] (SyS_init_module) from [<8000e660>] (ret_fast_syscall+0x0/0x48)
 remoteproc0: 0.remoteproc-test is available
 remoteproc0: Note: remoteproc is still under development and considered experimental.
 remoteproc0: THE BINARY FORMAT IS NOT YET FINALIZED, and backward compatibility isn't yet guaranteed.
 remoteproc0: Direct firmware load failed with error -2
 remoteproc0: Falling back to user helper
root@Avnet-Digilent-ZedBoard-2014_2:~#  remoteproc0: registered virtio0 (type 7)
random: nonblocking pool is initialized

I have also changed the ipino from 6 to 8 as has been suggested for the newer kernels.

Thanks,
Josh

wjliang, 2015-10-09:
> <80015658> <http://unwind_backtrace> from <800119b4>
> <800119b4> <http://show_stack> from <80490688>
> <80490688> <http://dump_stack> from <8048eda0>
> <8048eda0> <http://print_usage_bug> from <8005e6bc>
> <8005e6bc> <http://mark_lock> from <8005f760>
> <8005f760> <http://__lock_acquire> from <80061030>
> <80061030> <http://lock_acquire> from <80495d68>
> <80495d68> <http://_raw_spin_lock> from <8021a374>
> <8021a374> <http://gic_set_affinity> from <8021a3fc>
> <8021a3fc> <http://gic_set_cpu> from <7f093470>
> <7f093470> from <80273114> <http://platform_drv_probe+0x18/0x48>
> <80273114> <http://platform_drv_probe> from <80271958>
> <80271958> <http://driver_probe_device> from <80271c24>
> <80271c24> <http://__driver_attach> from <8026ff14>
> <8026ff14> <http://bus_for_each_dev> from <80270ffc>
> <80270ffc> <http://bus_add_driver> from <8027223c>
> <8027223c> <http://driver_register> from <800088b8>
> <800088b8> <http://do_one_initcall> from <80086070>
> <80086070> <http://load_module> from <80086770>
> <80086770> <http://SyS_init_module> from <8000e660>

Hi Josh,

I haven't tried OpenAMP with 3.14.2 kernel. but I have tried with 3.17
kernel in PetaLinux 2014.4 (tag openamp_zynq_kernel_3.17
<https://github.com/OpenAMP/open-amp/releases/tag/openamp_zynq_kernel_3.17>)

about the IPINO, have you check /proc/interrupts to see if 6 and 8 are
occupied? I have tried 15 and 14.

Best Regards,
Wendy

joshkern, 2015-10-09:
I'm going through the same steps on 2014.4 just for debugging purposes, but ultimately I want to use 2014.2 since I'm using that version for all my other tools (Vivado, SDK, etc.).

I have checked /proc/interrupts, and before adding the remoteproc node on the device tree, the only IPIs that are occupied are IPI1 through IPI7

joshkern, 2015-10-09:
@wjliang About the original comment you made on this thread. You said if using Petalinux 2014.4+ we should use the "zynq_remoteproc" that's built into the kernel. Isn't that the same driver as "zynq_remoteproc_driver" found in <project root>/components/modules/ ? The only other remoteproc driver is the generic one, just "remoteproc". Is this the one that you meant?

wjliang, 2015-10-11:
> as "zynq_remoteproc_driver" found in /components/modules/ ? The only other

Hi Josh,

Since it is very confusing to have the remoteproc driver in multiple
places, it is better to just have one place to hold the driver. The
zynq_remoteproc_driver in the
https://github.com/OpenAMP/open-amp/blob/master/apps/samples/master/linux/kernelspace/zynq_remoteproc_driver/zynq_remoteproc_driver.c
is the same as the one in drivers/remoteproc/zynq_remoteproc.c in the
kernel in 2014.4 PetaLinux.

However, the zynq_remotepro_driver in github is obsolete, and it will be
removed. I hope this can reduce your confusion.

Best Regards,
Wendy

wjliang, 2015-10-12:
Hi Josh,

Since it is very confusing to have the remoteproc driver in multiple
places, it is better to just have one place to hold the driver. The
zynq_remoteproc_driver in the
https://github.com/OpenAMP/open-amp/blob/master/apps/samples/master/linux/kernelspace/zynq_remoteproc_driver/zynq_remoteproc_driver.c
is
the same as the one in drivers/remoteproc/zynq_remoteproc.c in the kernel
in 2014.4 PetaLinux.

However, the zynq_remotepro_driver in github is obsolete, and it will be
removed. I hope this can reduce your confusion.

Best Regards,
Wendy

joshkern, 2015-10-12:
I'm still confused. If those two drivers are the same, how is one obsolete and the other is valid?

wjliang, 2015-10-13:
Hi Josh,

the zyqn remoteproc driver in the github will not be maintained or updated
and will be removed. Only the one in the
https://github.com/xilinx/linux-xlnx will be maintained and updated.

I should have said the zynq_remoteproc_driver in openamp github repo will
be obsolete. Sorry for the wording.

Best Regards,
Wendy

joshkern, 2015-10-13:
For anyone trying to follow the OpenAMP Framework Getting Started Guide using Petalinux 2014.4, make sure to use:

modprobe zynq_remoteproc firmware="/zc702evk/baremetal/matrix_multiply/firmware"

instead of

modprobe zynq_remoteproc firmware="/lib/firmware/zc702evk/baremetal/matrix_multiply/firmware"

Took me a couple days of experimentation/kernel debugging to figure out the simple change