OpenAMP/open-amp #25: Linux: select() call does not work with char device created by user_dev_driver

brummer-simon, 2017-03-07:
Hi Everyone:

I would like build some asynchronous processing logic with OpenAMP. I intend to use select() upon an opened file descriptor to block until there is data available or a timeout expired. The Problem is that the select call doesn't block then used with OpenAMPs file descriptor. The cause of this might be the missing poll() functionality. 

I would really love to seem this feature soon.

wjliang, 2017-03-08:
Thanks to raise missing file operations from OpenAMP. Would you also  like to contribute to adding these operations? 

I expect they could be  similar to the file operations defined in https://github.com/OpenAMP/open-amp/blob/master/lib/proxy/rpmsg_retarget.c. And on Linux side, will also need to add implementation to parse the select() and  poll().

brummer-simon, 2017-03-08:
Personally I would like to do that, the problem is I use OpenAMP in  a commercial project and I am not sure that I don't get the time to do that. I just wanted to raise awareness.

arnopo, 2021-04-28:
Close the issue as staled since for long time, please reopen it or open a new one if needed.

closed 2021-04-28 by arnopo