OpenAMP/open-amp #26: Please make the OpenAMP API const correct

franzhollerer, 2017-06-16:
Nowadays const correctness should be self-evident, especially for library code. It makes the API more clear, gives a contract on which data will not be modified by the callee and helps the compiler to warn about violations. In total const correctness helps to improve usability, quality and maintainability of the code.

Unfortunately, I found that the OpenAMP implementation does not care about const correctness. I consider this is serious bug.

For example the *rpmsg_send_offchannel_raw()* has the following signature:

```c
/**
 * This function sends rpmsg "message" to remote device.
 *
 * @param rp_chnl - pointer to rpmsg channel
 * @param src     - source address of channel
 * @param dst     - destination address of channel
 * @param data    - data to transmit
 * @param size    - size of data
 * @param wait    - boolean, wait or not for buffer to become
 *                  available
 *
 * @return - status of function execution
 *
 */

int rpmsg_send_offchannel_raw(struct rpmsg_channel *rp_chnl, uint32_t src,
			      uint32_t dst, char *data, int size, int wait);

```

Hopefully, the *data to transmit* are not modified by the function. Therefore it should be qualified as const. I am not sure if *rp_chnl* is modified by the function. This is hard to figure out at a glace (which is a good example why const correctness important).

Assuming that *rp_chnl* is constant too, the correct function prototype would be:

```c
int rpmsg_send_offchannel_raw(const struct rpmsg_channel *rp_chnl, uint32_t src,
			      uint32_t dst, const char *data, int size, int wait);
```

Another example is the function *rpmsg_get_buffer_size()*:

```c
/**
 * rpmsg_get_buffer_size
 *
 * Returns buffer size available for sending messages.
 *
 * @param channel - pointer to rpmsg channel
 *
 * @return - buffer size
 *
 */
int rpmsg_get_buffer_size(struct rpmsg_channel *rp_chnl):

```
It is obvious that the function will not change the passed parameter. Thus the correct function prototype is:

```c
int rpmsg_get_buffer_size(const struct rpmsg_channel *rp_chnl):
```

 I think this is essential therefore I ask you to make the OpenAMP API const correct.

wjliang, 2017-07-06:
good comment, let me check the code and add the const

wjliang, 2017-12-13:
"const" has been added to the APIs

closed 2017-12-13 by wjliang