OpenAMP/open-amp #27: elf_loader is not compatible with C++ executable

alexian79, 2017-08-04:
Hi I am not sure but following code in elf_loader_load_sections prevents loading C++ constructor initialization section
			/* Check if the section is part of runtime and is not section with
			 * no-load attributes such as BSS or heap. */
			if ((current->sh_type & SHT_NOBITS) == 0) {

sh_type for that kind of section is #define     SHT_INIT_ARRAY          14
which is 0xE in hex
BR, Alex.

wjliang, 2018-01-26:
Do you mean C++ constructor initialization section will have SHT_NOBITS (that is no data) ?

Thanks,
Wendy

alexian79, 2018-01-27:
yes exactly

wjliang, 2018-02-15:
I am thinking to use load loadable segment instead of check the details of each section.

wjliang, 2018-02-21:
please check this issue https://github.com/OpenAMP/open-amp/issues/40 for the change of remoteproc to load a firmware, will load segments instead of sections, will it work for your case?

wjliang, 2018-09-24:
This issue should be solved in the latest OpenAMP code.

closed 2018-09-24 by wjliang