OpenAMP/open-amp #29: Provide extern "C" declarations for the case the code is mixed with C++

franzhollerer, 2017-09-12:
Today it is common pactice to add the code shown below to C header files in order to allow their use from C++:

```c
#if defined __cplusplus
extern "C" {
#endif
:
:
#if defined __cplusplus
}
#endif

```

I have seen that the files under the *obsolate* directory follow this practise. Unfortunately, the files under lib/include/openamp don't. This gives some inconvenience if you want to use the open-amp implementation with C++.

Please consider this is as improvement request for future versions.

wjliang, 2017-09-13:
thanks for the suggestion. just we haven't tested OpenAMP with C++, you can make the change in the include/openamp/open-amp.h and the include/openamp/hil.h. if it works for you,  you are welcomed to the send the patch.

Thanks,

wjliang, 2018-02-13:
extern "C" linkage specification is added to the header if cpp compiler is used

closed 2018-02-13 by wjliang