OpenAMP/open-amp #37: Why remoteproc_boot doesn't use an entry_point address but uses a load address?

tyokota-eforce, 2018-01-21:
Hi , all.
Now  remoteproc_boot() uses  remoteproc_get_load_address() for a remote CPU boot address.
But I think it should use remoteproc_loader_retrieve_entry_point()  instead of  remoteproc_get_load_address().

wjliang, 2018-01-26:
You are right, it should use remoteproc_loader_retrieve_entry_point()

it looks like you want to use remoteproc to load firmware.
Just let you know that I am working on fixing the remoteproc to bring up the remote system.
Here is some document: https://github.com/wjliang/open-amp/blob/decouple-rproc-rpmsg/docs/design.md
and https://docs.google.com/document/d/1j8m543CoJE_DyM0WxU9FEpHSGUb5eR3TemKgaJ5fy5s/edit#heading=h.49r49nzgt64s for loading firmware from storage.

Could you share your use case?

tyokota-eforce, 2018-02-01:
Thank you to share your document.
Our use case improve linux remote amd userspace rpmsg.
I'm already successed linux remote boot and userspace rpmsg on Zynq7000 from our proprietary rtos.

wjliang, 2018-02-02:
Good to know OpenAMP works for you. just want to know "userspace rmsg" does it mean you use the OpenAMP library implementation from userspace? That is you don't need to have rpmsg kernel space driver such as the rpmsg_user_dev_driver.

tyokota-eforce, 2018-02-05:
That's right. 
We use open-amp petalinux 2017.2 package on Zynq7000 in linux side.
And we ported open-amp library for our rtos.
It works fine.

wjliang, 2018-02-21:
I have created an issue to change how the remoteproc loads a firmware, please take a look at please check this issue https://github.com/OpenAMP/open-amp/issues/40, the required change reported in this issue will be fixed in the change of issue #40

wjliang, 2018-09-24:
the latest OpenAMP code use ELF entry address

closed 2018-09-24 by wjliang