OpenAMP/open-amp #43: Can I use OpenAMP without running linux?

oska874, 2018-03-19:
Can I use OpenAMP without running linux?
Is it possible to runing RTOS/bare-metal on all cpu cores without linux?

wjliang, 2018-03-19:
It is possible, you can write RPMsg master (who starts the communication) RTOS/bare-metal for your platform, you can try https://github.com/OpenAMP/open-amp/blob/master/apps/echo_test/echo_test.c.
I haven't tested it, but the OpenAMP implementation is supposed to work across OSes.

wjliang, 2018-03-20:
Sounds good, what changes of OpenAMP library did you make? Are you going to post patches?

alexian79, 2018-03-20:
We can discuss this changes in this topic. My changes were done to lib from xilinx sdk v1.3. I will explain them here a bit later.

PetrLukas, 2018-03-21:
If you are looking for light weight implementation compatible with RPMSG protocol then RPMSG-Lite could be a good option. [https://github.com/codeauroraforum/rpmsg-lite](url)

alexian79, 2018-03-21:
For now I couldn't find anything better than to post my old playground Xilinx SDK directory(targeted Microzed) as an example.
[https://github.com/alexian79/bm2bm_openamp_example](url)
It contains  modified libmetal and openamp libs. Probably you will need to add this dir to SDK global modules search path. Libmetal does not have any significant changes except C++ guards ware broken in original one. In openamp main hack was to simulate/hack Virtual device interrupt to kick off slave ready event(it is using ipi vector 0). Additionally some cache flushing in arm9 after loading slave image.
 In example projects from SDK templates, I think I did some resource map and init polishing to make it work. I dont think this hacks is a good way to implement missing VIOD.

I tried to create branch derived from what I suppose were xilinx version 1.3 here as well, to simplify to see what changed.
 but in this one I did some modifications to rpc to allow messages more than rpc buffer size.
  [https://github.com/alexian79/open-amp](url)

I will be back on experimenting with open amp soon.

wjliang, 2018-03-22:
Thanks for your sharing
I am not able to access  https://github.com/alexian79/bm2bm_openamp_example nor https://github.com/alexian79/open-amp.
It looks like you have add some feature to allow message more than the buffer size.
It will be good that if you can send a pull request .
Thanks,
Wendy

wjliang, 2018-03-26:
I am able to see your repos.

I notice that in the zynq_a9 remoteproc driver, set the virtio status to DRIVER_OK in the interrupt handler.
Is this updated driver used for master only? or both?
I think the status should be set up DRIVER_OK when rpmsg device finish the shared buffers intialiazation.

alexian79, 2018-03-27:
This is that hack I was talking about.
I am not sure about right terminology in this case. Only one part is using it. 
it registers this handler in  lib/rpmsg/remote_device.c at line 155 if role is not RPMSG_REMOTE
if (rdev_loc->role == RPMSG_REMOTE) {
		/*
		 * Since device is RPMSG Remote so we need to manage the
		 * shared buffers. Create shared memory pool to handle buffers.
		 */
		shm = hil_get_shm_info(proc);
		rdev_loc->mem_pool =
		    sh_mem_create_pool(shm->start_addr, shm->size,
				       RPMSG_BUFFER_SIZE);

		if (!rdev_loc->mem_pool) {
			return RPMSG_ERR_NO_MEM;
		}
	}
	else
	{
		**proc->vdev.intr_info.dev = (struct metal_device *)virt_dev;
		hil_set_vdev_ipi(proc, 0, 0, NULL);**
	}

wjliang, 2018-04-02:
You can take a look at rpmsg_start_ipc(), if the remote is "remote", it will set the VIRTIO_CONFIG_STATUS_DRIVER_OK:
```
if (rdev->role == RPMSG_REMOTE) {
                virt_dev->func->set_status(virt_dev,
                        VIRTIO_CONFIG_STATUS_DRIVER_OK);
                status = rpmsg_rdev_notify(rdev);
        }
```

alexian79, 2018-04-04:
I think this was exactly the place where it stuck. Because notify will try to trigger ipi interrupt 0. and there was no handler for that. And I needed to save virt_dev context somewhere as well then it will be triggered.

It was constantly looping in this function 
 int rpmsg_rdev_remote_ready(struct remote_device *rdev)

waiting for this condition 
if (status & VIRTIO_CONFIG_STATUS_DRIVER_OK) {

wjliang, 2018-04-04:
However, as they are using shared memory, as long as the master side update the vdev status to DRIVER_OK. The remote side should get it.
You can check on the master side to see if the status is set in the vdev shared memory, and then can connect to the slave side to see if it can see the updated shared memory.

Or did you it is possible that the vdev is in different location on the master side and the slave side. You can make sure the ".resource_table" section is in the same location for both master and slave, then you should not need the hack

alexian79, 2018-04-04:
Thanks for the suggestion, I will double check that. I think in my case it was 2 copies of the device, not shared memory. Maybe it is related to that in remoteproc_init  function I inserted line
		rproc->proc = proc;
Do I need to create proc for each processor?

wjliang, 2018-04-05:
i am not sure what you mean by two copies of the device, you have fixed the bug in the remoteproc_init rproc->proc = proc which is already fixed in the upstream.
as you are using remoteproc_init, did you load the other firmware from the master side? in this case, looks like the resource table will be retrieved from the remote firmware. But you can check if you have mapped the resource table memory as non-cacheble memory.

diegosueiro, 2018-08-22:
@oska874 ,

Zephyr has support and a sample app for IPC, using OpenAMP, between a Cortex M0 and Cortex M4:
https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/subsys/ipc/openamp

It will compile the binaries for both cores and the M4 will be responsible for booting, loading and start the M0.

arnopo, 2021-04-28:
close the issue as it is probably no longer relevant, please reopen it or open a new one if needed.

closed 2021-04-28 by arnopo