OpenAMP/open-amp #44: How to clear/free/flush virtio buffer ring?

stahlstngel, 2018-03-28:
I am using rpmsg to communicate between a linux (A9) and a FreeRTOS (M4) on an imx6sx through a kernel module with ioctl on a device file on devfs.

The communication works…sending and receiving is fine, but when I reload a FreeRTOS image on the M4 side…all the previous send rpmsg messages will be received from the M4 again. Afterwards I can send and receive new messages. But it seems that the virtio buffer ring in the shared memory stays even I run "rmmod" the kernel module.

Is there a way to clear, free or flush the shared memory to communicate via IPC? I would prefer to flush the within my loadable kernel module.

wjliang, 2018-03-29:
I don't think the rmmod will clean the shared memory, it depends on where you define your shared memory, if the vring shared memory is allocated from DMA coherent, the remoteproc core should free the memory when you shutdown the processor. however, the content in the shared memory area will get clean.
But why do you also want to clean the shared memory content?

stahlstngel, 2018-03-29:
In my case the bootloader (UBOOT) defines the memory that is used for rpmsg communication.

My problem is that when I load a new M4 image without to restart the imx6sx the new image will receive the messages that was send to the old M4 image. So it seems to be that all rpmsg communication is held in the shared memory, even if a new M4 image is loaded (without imx6sx restart).

Therefore I would like to clean the shared memory at a "rmmod" so that a new image will not receive the rpmsg messages that was send to the old M4 image.

Can you confirm that the all rpmsg messages will held in the shared memory?

wjliang, 2018-04-02:
The shared memory in vring buffer will not be cleared. However, the vrings themselves should have been reinitialized if you reload the image and restart the communication. (that is even the RPMsgs are still in the shared memories, but as long as the vrings' got reinitialized,  it should be fine) unless your implementation in kernel side doesn't do that.

stahlstngel, 2018-04-16:
Indeed. It is needed to reinitializes the vring buffers when a new Image is loaded.

So far this issue seems more to be only relevant for development. For this I will close this issue as solved and say thanks to wjliang.

Have a nice day.

closed 2018-04-16 by stahlstngel

srpatcha, 2020-09-21:
Is there way that we can reinitialize RPC module to communicate when remote processor(M4) is rebooted ???

arnopo, 2020-09-21:
@srpatcha 
Please could you create a new issue instead using this closed one?
Thanks!