OpenAMP/open-amp #92: conflicting types for ‘sleep’

duane-etagen, 2018-05-25:
```
open-amp/apps/func_test_suite/func_test_suite.c:25:13: error: conflicting types for ‘sleep’
 static void sleep();
             ^
In file included from libmetal-install/usr/local/include/metal/system/linux/sys.h:58:0,
                 from libmetal-install/usr/local/include/metal/sys.h:111,
                 from libmetal-install/usr/local/include/metal/dma.h:47,
                 from open-amp/lib/include/openamp/virtqueue.h:37,
                 from open-amp/lib/include/openamp/virtio.h:34,
                 from open-amp/lib/include/openamp/rpmsg_core.h:35,
                 from open-amp/lib/include/openamp/rpmsg.h:39,
                 from open-amp/lib/include/openamp/open_amp.h:33,
                 from open-amp/apps/func_test_suite/func_test_suite.c:8:
/usr/include/unistd.h:447:21: note: previous declaration of ‘sleep’ was here
 extern unsigned int sleep (unsigned int __seconds);
                     ^
****
```
Compiling with `-DWITH_FUNCTION_TEST_SUITE=ON` from cmake.
OS is 
```
$ uname -a
Linux *** 4.13.0-43-generic #48~16.04.1-Ubuntu SMP Thu May 17 12:56:46 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```
```
$ gcc --version
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.9) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
Is it somehow unexpected to have `sleep(unsigned int)` defined here?

duane-etagen, 2018-05-25:
Also turns out that `sleep()` is never used. Left over debugging?

wjliang, 2018-05-25:
> Also turns out that sleep() is never used. Left over debugging?
It was the left over as previously we use sleep() in the for() loop when waiting for next request, later, we use hil_poll() to wait, and thus, there is no need to sleep in the app. the sleep() is left over, will be removed.

arnopo, 2020-12-03:
Seems solved since a long time.  close it

closed 2020-12-03 by arnopo