OpenAMP/open-amp #93: Linux user space apps don't connect

duane-etagen, 2018-05-25:
Compiled on Ubuntu 16.04. `echo_testd-shared` and `echo_test-shared` don't connect. 

```
$ sudo LD_LIBRARY_PATH=usr/local/lib/ usr/local/bin/echo_testd-shared
metal: debug:     Failed fread /dev/urandom
metal: debug:     added page size 4096 @/tmp
metal: debug:     added page size 2097152 @/dev/hugepages
metal: debug:     registered platform bus
metal: debug:     registered pci bus
Waiting for connection on /tmp/openamp.event.0
```

In another shell
```
$ sudo LD_LIBRARY_PATH=usr/local/lib/ usr/local/bin/echo_test-shared
metal: debug:     Failed fread /dev/urandom
metal: debug:     added page size 4096 @/tmp
metal: debug:     added page size 2097152 @/dev/hugepages
metal: debug:     registered platform bus
metal: debug:     registered pci bus
 1 - Send data to remote core, retrieve the echo and validate its integrity ..
Waiting for connection on /tmp/openamp.event.0
```

And ... nothing happens. 

I can repeat this for `matrix_multiply` as well.

I followed the instructions for building. No additional changes. 

I've run the tests for `libmetal` and they pass. (`make test`)

Thoughts? Where should I be looking?

wjliang, 2018-05-25:
```
sudo LD_LIBRARY_PATH=usr/local/lib/ usr/local/bin/echo_test-shared 1
```
the `echo_testd-shared` runs as the server, and the `echo_test-shared` runs as a client, the TX channel in client is the RX in server. `1` is the index of remote processor defined in `apps/system/linux/machine/generic/platform_info.c`, if you don't pass index it will always use remote `0`. The demo expect the tests run on different remotes:
```
sudo LD_LIBRARY_PATH=usr/local/lib/ usr/local/bin/echo_testd-shared
sudo LD_LIBRARY_PATH=usr/local/lib/ usr/local/bin/echo_test-shared 1
```

duane-etagen, 2018-05-25:
Thank you! That works much better. I totally missed the `1`. I'll copy and paste next time.

Closing.

closed 2018-05-25 by duane-etagen