OpenAMP/open-amp #95: EWARM warnings

rahmanih, 2018-06-01:
Hi,
Using the openAMP with IAR EWARM there are the following warnings:
`
Warning[Pe223]: function "metal_generic_dev_sys_open" declared implicitly OpenAMP\libmetal\lib\device.c 148 
`
<br>
`
Warning[Pe223]: function "metal_sys_io_mem_map" declared implicitly OpenAMP\libmetal\lib\system\generic\device_generic.c 51 
`
<br>
`
Warning[Pe223]: function "metal_sys_init" declared implicitly OpenAMP\libmetal\lib\init.c 47 
`
<br>
`
Warning[Pe223]: function "metal_sys_finish" declared implicitly OpenAMP\libmetal\lib\init.c 56 
`
<br>
`
Warning[Pe223]: function "metal_machine_io_mem_map" declared implicitly OpenAMP\libmetal\lib\system\generic\io_generic.c 50 
`
<br>
`
Warning[Pe223]: function "metal_sys_io_mem_map" declared implicitly OpenAMP\libmetal\lib\io.c 48 
`
<br>
`
Warning[Pe223]: function "sys_irq_save_disable" declared implicitly OpenAMP\libmetal\lib\system\generic\irq.c 258 
`
<br>
`
Warning[Pe223]: function "sys_irq_restore_enable" declared implicitly OpenAMP\libmetal\lib\system\generic\irq.c 266 
`
<br>
`
Warning[Pe111]: statement is unreachable  OpenAMP\open-amp\lib\rpmsg\remote_device.c 379 
`
<br>
`
Warning[Pe111]: statement is unreachable  OpenAMP\open-amp\lib\remoteproc\rsc_table_parser.c 163 
`
<br>
`
Warning[Pe223]: function "metal_shmem_open_generic" declared implicitly OpenAMP\libmetal\lib\system\generic\shmem_generic.c 41 
`
<br>

Is it possible to fix them?
regards
Haithem.

wjliang, 2018-06-01:
Could you share your compilation flags and which processor are you building? you have quite some "declared implicitly" warnings. e.g.
```
Warning[Pe223]: function "metal_sys_io_mem_map" declared implicitly OpenAMP\libmetal\lib\io.c 48
```
But they are declared in the generic/io.h.
And I also see you have some new `_generic.c` files, did you update the source code?

Have you tried to compile with the unchanged master branch?

Best Regards,
Wendy

wjliang, 2018-06-20:
Is this still an issue? Can I close it?

rahmanih, 2019-03-19:
it is my mistake!

closed 2019-03-19 by rahmanih