OpenAMP/open-amp #126: Using open-amp for M4

andi32, 2018-08-03:
Hi,

i just want to use RPMsg baremetal on the slave side (Linux side is already running). I am quite confused what components i need to port. In your documentation of open-amp there is written to port the platform.c.h files. But in libmetal  there is written I need to port alloc,cache etc.. So I need to to both?

General Question:
How do both systems (master and slave) knows what kind of hardware layer (interrupt base, mailboxes etc,) they use for communication? Is there a common file/header for checking compatibility? 

Thanks Andreas

wjliang, 2018-08-04:
> Hi,
>
> i just want to use RPMsg baremetal on the slave side (Linux side is already running). I am quite confused what components i need to port. In your documentation of open-amp there is written to [...]

platform.c provides examples on platform specific OpenAMP porting: such as notification. 
libmetal provides cross OSes abstractions to access hardware (I/O, memory, cache), you can refer to https://github.com/OpenAMP/libmetal/blob/master/lib/system/generic/template/sys.c to port libmetal for M4

> 
> General Question:
> How do both systems (master and slave) knows what kind of hardware layer (interrupt base, mailboxes etc,) they use for communication? Is there a common file/header for checking compatibility?

For libmetal, please refer to https://github.com/OpenAMP/libmetal/blob/master/lib/system/generic/template. Libmetal doesn't contains driver, but it provides hardware access + some other primitives across different OSes, so that user can develop cross OS drivers/libraries/apps/ For OpenAMP, it is to implement the driver required by virtio and remoteproc. If you take a look at https://github.com/OpenAMP/open-amp/blob/master/apps/machine/zynqmp_r5/platform_info.c, it implements platform specific remoteproc operations, and platform specific system, remoteproc and virtio initialization.

Best Regards,
Wendy

> 
> Thanks Andreas

andi32, 2018-08-06:
HI, 
sorry but I am still confused. Do I need to make adaption to OpenAMP (platform...) and libmetal(sys...)?

Andreas

andi32, 2018-08-07:
Hi,
before OpenAMP using the libmetal library there was a really good documentation (open_amp.ref) how to start and how it works. But unfortunately the documentation wasn't updated and so it's really hard to for beginners to get started. It would be nice if you can fix this in the future. 

Andreas

arnopo, 2020-12-03:
The document has been removed by pull request: #174,  
issue https://github.com/OpenAMP/open-amp/issues/172 list some remaining stuff

closed 2020-12-03 by arnopo