OpenAMP/open-amp #135: OpenAmp on Microblaze

Shivrajatria, 2018-09-19:
I am trying to port the example mentioned in **UG1186** onto Microblaze (Soft IP).
I work on zc102 board, I want to make communication between cortex 53 and Microblaze (PL->Soft IP).
I tried googling about it but found none.
If anyone has any Idea how to do that please help.

wjliang, 2018-09-20:
> I am trying to port the example mentioned in **UG1186** onto Microblaze (Soft IP).
> I work on zc102 board, I want to make communication between cortex 53 and Microblaze (PL->Soft IP).
> I tried googling about it but found none.
> If anyone has any Idea how to do that please help.

if you want to load MicroBlaze firmware, you will need to create a microblaze remoteproc kernel driver, you can refer to any driver in the linux  kernel drivers/remoteproc directory.
If you just care about the communication, the simplest way is to try OpenAMP in userspace example, which is also documented in ug1186

closed 2019-02-08 by ShivrajJuvekar

altair5683, 2019-05-15:
@ShivrajJuvekar Hello. Do you have idea how to port OpenAMP to MicroBlaze? I am trying to solve the same port problem.