OpenAMP/open-amp #137: Communicate between multi  linux processes and rtos on same RPMsg channel?

tyokota-eforce, 2018-10-10:
We would like to communicate between two linux processes, these use different endpoints, and RTOS tasks by RPMsg on same channel.
Is this function  possible? and if it's possible what should we do?

`(Linux)  　　　　　　　　　　　　　　　 (RTOS)`
`[proc1(user app)]#ep1  -- [vring]   ---  [ RTOS task(app)#ep1]`
`[proc2(mng app)]#ep2 　　　　　　　　　  [ RTOS task(mng)#ep2]`

wjliang, 2018-10-11:
If you use Linux kernel RPMsg char driver, you can create two endpoints from userspace that is each process can create one endpoint, or you create your own RPMsg kernel driver then, you can expose two endpoints as files to userspace.

If you want to run the RPMsg implementation in OpenAMP in userspace, that is the openamp lib/rpmsg in userspace, it supports multiple threads and each threads use different endpoints. But it may not work with two independent Linux user applications.

tyokota-eforce, 2018-10-15:
Thank you very much for your suggestions.
I'll try them.

closed 2018-10-30 by wjliang