OpenAMP/open-amp #166: rpmesg_virtio: Statement unreachable

rahmanih, 2019-03-19:
Hi,
the function rpmsg_virtio_wait_remote_ready() is throwing the following warning with MDK-ARM compiler:
`
OpenAMP/open-amp/lib/rpmsg/rpmsg_virtio.c(236): warning:  #111-D: statement is unreachable
`
the function  implementation is as following:

```
	while (1) {
		status = rpmsg_virtio_get_status(rvdev);
		/* Busy wait until the remote is ready */
		if (status & VIRTIO_CONFIG_STATUS_NEEDS_RESET) {
			rpmsg_virtio_set_status(rvdev, 0);
			/* TODO notify remote processor */
		} else if (status & VIRTIO_CONFIG_STATUS_DRIVER_OK) {
			return true;
		}
		/* TODO: clarify metal_cpu_yield usage*/
		metal_cpu_yield();
	}

	return false;
}
```
with the **`while(1)`** loop the  **`"return false;"`** in indeed unreachable.

wjliang, 2019-05-25:
agree, the "return false;" should be removed.

rahmanih, 2020-02-17:
issue Fixed!

closed 2020-02-17 by rahmanih