OpenAMP/open-amp #172: Outdated documentation

DaveTrussel, 2019-05-01:
Looks like this document is quite old and describes a different API.
https://github.com/OpenAMP/open-amp/blob/master/docs/openamp_ref.pdf

wjliang, 2019-05-25:
The document has been removed by pull request: https://github.com/OpenAMP/open-amp/pull/174

ChrisYin, 2019-08-29:
README files in docs/apps are also out-dated.
```
obsolete/apps/echo_test/system/linux/kernelspace/rpmsg_user_dev_driver  if you want to run the echo_test app in Linux user space.
obsolete/system/linux/kernelspace/rpmsg_echo_test_kern_app  if you want to run the echo_test app in Linux kernel space.

```
There is no obsolete directory in the current version.

arnopo, 2024-02-28:
https://openamp.readthedocs.io/en/latest/ should answer to this issue

closed 2024-02-28 by arnopo