OpenAMP/open-amp #175: microblaze openamp IPI probleam

pigeonsoup, 2019-06-05:
Hi, I was trying to use microblaze to realize openamp echo-test example. I have checked https://github.com/OpenAMP/open-amp/issues/135, so I could only use it in linux userspace. I have already designed my vivado project(refer to https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18841793/Utilizing+PS+memory+to+execute+Microblaze+application+on+Zynq+Ultrascale?search_id=a14232a6-1b0e-40fa-acd1-3fd39a4c94fc), then lanched to SDK and build my microblaze echo-test application. On the other side, I got my petalinux project ready. However, when I started the test, it seems that microblaze side never got the IPI inerrupt message from linux, which means that linux side send a number and receive nothing from microblaze side.

Microblaze app have already finished rpmsg initialization and start polling, so I think it?did not get IPI interrupt.

Here is my console:

root@mb-test:~# echo_test-openamp
metal: info: metal_uio_dev_open: No IRQ for device 3ed40000.vring.
metal: info: metal_uio_dev_open: No IRQ for device 3ed40000.vring.
metal: info: metal_uio_dev_open: No IRQ for device 3ed80000.shm.
1 - Send data to remote core, retrieve the echo and validate its integrity ..
Remote proc resource initialized.

And here is my terminal(from my sdk echo-test project):

Starting application...
Try to init remoteproc resource

Init remoteproc resource succeeded
Waiting for events...

Here is my microblaze code.

pigeonsoup, 2019-06-05:
I do not know how to attach my code, it seems  "Attach files by dragging & dropping, selecting or pasting them"below does not work

edmooring, 2019-10-04:
This is probably a device tree issue. I'm not familiar with the Microblaze device tree setup, but please attach your device tree, or put the relevant snippets for the uio devices in a comment.

arnopo, 2020-12-03:
Hi @pigeonsoup,
As this issue is stalled since a long time i'm closing it, please re-open it if necessary.

closed 2020-12-03 by arnopo