OpenAMP/open-amp #176: Why disable virtqueue callback function?

ChrisYin, 2019-09-04:
In file lib/rpmsg/rpmsg_virtio.c, line 587: it disables the callback function.

When running the echo examples, the remote core will not get any response when the master core runs virtqueue_kick function[because vq_ring_must_notify wiil return false].

```
	/*
	 * Suppress "tx-complete" interrupts
	 * since send method use busy loop when buffer pool exhaust
	 */
	virtqueue_disable_cb(rvdev->svq);
```

```
/**
 * virtqueue_kick - Notifies other side that there is buffer available for it.
 *
 * @param vq      - Pointer to VirtIO queue control block
 */
void virtqueue_kick(struct virtqueue *vq)
{
	VQUEUE_BUSY(vq);

	/* Ensure updated avail->idx is visible to host. */
	atomic_thread_fence(memory_order_seq_cst);

	if (vq_ring_must_notify(vq))
		vq_ring_notify(vq);

	vq->vq_queued_cnt = 0;

	VQUEUE_IDLE(vq);
}
```

edmooring, 2019-10-04:
@wjliang Do you remember the reasoning behind this change?

arnop2, 2019-10-11:
@edmooring,  @ChrisYin 
Please refer to https://github.com/OpenAMP/open-amp/pull/147
if VIRTIO_RING_F_EVENT_IDX  is not set  not interrupt is generated, this can be used for rpmsg buffer polling mode as described here:
https://elixir.bootlin.com/linux/v4.3/source/include/uapi/linux/virtio_ring.h#L50
Anyway regarding code, something seems wrong in the implementation. We compare a bit-field with a bit index in vq_ring_enable_interrupt and vq_ring_must_notify:
	`if (vq->vq_dev->features & VIRTIO_RING_F_EVENT_IDX) {`

Waiting a clean fix, a workaround could be to add non null value for dfeatures an gfeatures

arnop2, 2019-10-17:
My apologize, seems i made confusion between the kernel and OpenAMP VIRTIO_RING_F_EVENT_IDX definition...on OpenAMP the  VIRTIO_RING_F_EVENT_IDX is a mask = 1<< 29)
By configuring dfeature in resource table with this flag, i have similar issue. The Linux kernel sent a kick only on the first message. 
@ChrisYin could you check that don't activate this Flag in your resource table?
Thanks

xiaoxiang781216, 2019-11-09:
@ChrisYin your understand is wrong:
1.vq_ring_must_notify in virtqueue_kick test whether the peer want to receive the interrupt.
2.virtqueue_disable_cb just notify the peer don't interrupt me after you finish process my request.
The first flag is decided by peer, not by virtqueue_disable_cb.

arnopo, 2020-12-03:
@ChrisYin
As this issue is stalled since long time, i consider that answers have help you to solve your issue. I close the issue. Pleas re-open it if the point is not close.

closed 2020-12-03 by arnopo