OpenAMP/open-amp #177: Load_FW demo on ZCU102

ajc112, 2019-09-06:
I'm attempting to get the load_fw demo to work on a ZCU102 eval board, and having much difficulty. Ultimately the goal is to have the APU running vxWorks and bring either RPU core out of reset and run arbitrary code, but right now I'm looking to understand what is provided here.

What steps would be needed to run the demo as is on the ZCU102 eval board?

Is there any further documentation? It's unclear to me what the target platform is or how to provide the .elf file to be loaded for the load_fw as provided.

edmooring, 2019-10-04:
The load_fw demo is specifically aimed at bringing up a simple executable (for example "Hello world") on the APU from the RPU. This is the opposite direction from what you want. If you were running Linux on the APU, you could simply use the remoteproc framework to reset the RPU, load the code, and run it.
I think, but don't know for sure, that VxWorks supports something similar to remoteproc.

bentheredonethat, 2020-02-05:
Hi,

Sorry for the late response. If you are still interested in loading RPU from the APU, consider looking at https://github.com/OpenAMP/open-amp/blob/master/apps/examples/load_fw/zynqmp_rpu_lcm_rproc_example.c. Additionally, when building the Open-AMP, the method by which you should build to then run this is to add -DLOAD_FW  and for the target (in this case RPU0 on ZU+) also add the target -DLOAD_FW_TARGET=NODE_RPU_0 

Hope this helps,
Ben

arnopo, 2020-12-03:
Hi @ajc112 ,
No activity on this issue for a long time, I consider that the answers have helped you to solve your problem. I close the subject. Please reopen it if the point is not closed.

closed 2020-12-03 by arnopo