OpenAMP/open-amp #179: A35 to slave in Linux

srpatcha, 2019-09-17:
How do I make changes in RPMSG driver for moving A35 to act as a slave in Linux?
Since we are moving M4 as a master in RTOS.

arnopo, 2020-09-21:
You are right the Linux Kernel  implementation only supports the Master mode. I'have never seen any Linux kernel  patchset to support the slave mode...
An alternative used by some developers to get around this limitation is to use the OpenAMP library in Linux Userland instead of using the Linux kernel RPMsg.
One question: could you explain why you want to use the RPMsg master mode for the RTOS?

srpatcha, 2020-09-21:
Trying to control the Remote Processor and to reinitialize RPMSG communication. 
"1. Provides Life Cycle Management and Inter Processor Communication capabilities for the management of remote compute resources and their associated software contexts. RPROC --> stop, restart, suspend?"
Would you explain the strategies of RPMSG reinitialize stages to approach in Linux?
I see an Error when IMX-M(RTOS) is rebooted and IMX-A(Linux) is waiting to **reinitialize-RPC** on imx_4.14.98:
[  133.783689] Wait for remote ready timeout, **assume it's dead.**
[  133.790257] virtio_rpmsg_bus virtio0: rpmsg host is online
[  133.821206] virtio_rpmsg_bus virtio1: rpmsg host is online
[  133.827268] Patition3 reset!
[  134.386933] Unable to handle **kernel NULL pointer** dereference at virtual address 00000324

arnopo, 2020-09-21:
> Trying to control the Remote Processor and to reinitialize RPMSG communication.
> "1. Provides Life Cycle Management and Inter Processor Communication capabilities for the management of remote compute [...]

I would suggest to contact  NXP support, because The start/stop/suspend/restaure/crash strategies are platform dependent.
you can also have a look In Linux remoteproc framework to the "recover" mechanism , to handle the co-processor crash. You probably have to check that remoteproc IMX driver supports the detection of the RTOS crash/reboot.

Here probably not correlation with the OpenAMP library (even less if the OpenAMP lib is not used and replaced by the NXP  rpmsg lib on the IMX-M)...

arnopo, 2020-12-03:
Hi @srpatcha  ,
No activity on this issue for a long time, I consider that the answers have helped you to solve your problem. I close the subject. Please reopen it if the point is not closed.

closed 2020-12-03 by arnopo

FirstLoveLife, 2023-07-03:
@arnopo Hi, rpmsg newbie here. What should linux kernel do if it wants to support slave mode?