OpenAMP/open-amp #182: Can't RPMsg remote send

kbumsik, 2019-11-08:
Hi, I'm using [ST vendor's OpenAMP port](https://github.com/STMicroelectronics/STM32CubeMP1/tree/3739122ec3281ffe18356774ffc06a7fbcf6df98/Middlewares/Third_Party/OpenAMP) but I think it is applied to the upstream as well so I'm posting a question here.

While working with RPMsg I found that the remote cannot send message using `rpmsg_send()` until the master sends any data first because `ept->dest_addr` is set to `RPMSG_ADDR_ANY` in the vendor. `ept->dest_addr` is corrected in [the fist `rpmsg_virtio_rx_callback`](https://github.com/OpenAMP/open-amp/blob/e1af16302c84934068573b7a786f0d36f11576d0/lib/rpmsg/rpmsg_virtio.c#L404):

```
if (ept->dest_addr == RPMSG_ADDR_ANY) {
    /*
     * First message received from the remote side,
     * update channel destination address
     */
    ept->dest_addr = rp_hdr->src;
}
```
[This looks like it's OpenAMP's design decision](https://github.com/OpenAMP/open-amp/blob/7b32a1b72cdc8866eea0803f683c35af01814a56/docs/img/coprocessor-rpmsg-ns.png) that blocks sending data from the remote until the master sends data.

I don't know if my thoughts is correct. If it is how to workaround this limitation? Should the master send any data when an endpoint is created, or I should figure out the correct local endpoint's destination address when creating the endpoint?

arnopo, 2019-11-08:
Hi @kbumsik,
yes it is a design decision. To establish a RPMsg channel both side has to be ready to communicate.
when a new service is created, The receiver has to send a first message to communicate is own address. So you have at least to send an  acknowledge dummy message.
if you are using the TTY over rpmsg, you have to send a first message from your Linux application.

kbumsik, 2019-11-08:
@arnopo 
Thank you very much for explanation.
> if you are using the TTY over rpmsg, you have to send a first message from your Linux application.

Does it also mean that the remote application should discard the first message manually if it is meant to be dummy too?

arnopo, 2019-11-08:
The rule in this ST use case is that the Linux application has to send the first message. this message will be received by the remote application. So it can be a dummy or a normal message. It depends of your implementation. So either you discard the first one or you can also implement something similar to the UART flow control:
- send "RTS 1" string when ready to receive ( this will be the first message
- send "RTS 0" string when you want to suspend the reception.

kbumsik, 2019-11-08:
I got it. Thank you for your reply!

I'm closing the issue since I got the best answer.

closed 2019-11-08 by kbumsik