OpenAMP/open-amp #183: OpenAMP on x86

XYEFA, 2019-11-13:
All of the OpenAMP examples I have seen are ARM specific. Is OpenAMP supported on x86 (Linux + RTOS).

xiaoxiang781216, 2019-11-15:
We test OpenAMP between two RTOS which run on Ubuntu as a hybrid simulator:
https://bitbucket.org/nuttx/nuttx/src/master/boards/sim/sim/sim/README.txt:rpserver section
Here is info about how the simulator work:
https://nuttx.org/doku.php?id=wiki:nxinternal:simulator

XYEFA, 2019-11-15:
Thank you for the answer - Do you have to rebuild Ubuntu or is OpenAMP or remoteproc included natively?

xiaoxiang781216, 2019-11-15:
> Thank you for the answer - Do you have to rebuild Ubuntu or is OpenAMP or remoteproc included natively?

No, but OpenAMP officially support two native instance(process) talk each other in Ubuntu.

closed 2020-04-29 by arnopo