OpenAMP/open-amp #185: rpmsg_ept_cb should optionally allow buffer to be held for extended processing

wmamills, 2019-12-19:
Currently the rpmsg callback is required to use the buffer only in the callback itself.
Although the default is fine for many applications if causes extra copies in other cases.

If the data in the buffer is needed for further processing or needs to be enqued to another task context, the data has to be copied (probably alloc, copy, and free later)

There should be some way to indicate to the rpmsg layer that the buffer will be held and freed at a later time.  A Bool return value could indicate this.  When it does come time to return the buffer, is the data pointer value enough or should there be a abstract handle?

xiaoxiang781216, 2019-12-19:
This feature(NXP provided) merge into mainline almost two years ago, but is removed in last year.
I don't know why the maintainer remove the feature without any discussion.
Then my team bring back this feature here more than six months:
https://github.com/OpenAMP/open-amp/pull/162
No any maintainer take the action really, I think OpenAMP community is almost stop functionality now.

arnopo, 2019-12-19:
> This feature(NXP provided) merge into mainline almost two years ago, but is removed in last year.
> I don't know why the maintainer remove the feature without any discussion.

As the main objective of restructuring OpenAMP was to significantly reduce the size of the code, it was decided to remove this functionality in a first step and to study its reintroduction  on demand. 
you convince me that we have to reintroduce it, thank for the PR.      

> Then my team bring back this feature here more than six months:
> #162
> No any maintainer take the action really, I think OpenAMP community is almost stop functionality now.

As you don't have all the context please don't give this kind of unconstructive  judgment. If you want to share your requirement and participate to the evolution of the OpenAMP project you are welcome (https://www.openampproject.org/about/). 

Concerning your PR itself:
Be sure that we plan to integrate your PR. But this could take some times as we have first  to released patches already integrated, before restarting a normal flow.

One blocking point today for its integration is the test. as you extend the API we need to test it and to ensure non regression in future. for this we need an application example that can run for test.
So to ease our work please provide the associated [application sample](https://github.com/OpenAMP/open-amp/tree/master/apps/examples). 

Another concerns is the footprint penalty associated.
We need to determine the impact of this additional code and decide if we should make it optional ( using compilation switch). But this point is minor...

xiaoxiang781216, 2019-12-20:
> As the main objective of restructuring OpenAMP was to significantly reduce the size of the code, it was decided to remove this functionality in a first step and to study its reintroduction on [...]

I can't find any link on this web that I can join the committee.
Sorry, my word may hurt you, but it is my real feeling about OpenAMP community.
I make a summary why we make all these change one month ago here:
https://groups.google.com/forum/#!topic/open-amp/VEWXUbNF1No
but nobody from the committee give any response.

> 
> Concerning your PR itself:
> Be sure that we plan to integrate your PR. But this could take some times as we have first to released patches already integrated, before restarting a normal flow.
>

When will this year offiical release announce?
According the project plan, OpenAMP should have two(April and October) official release in each year, but I don't see any release in this year yet.

And I guess that Wendy(the project leader) doesn't particent in this project anymore, is it right? I have many good technical conversation with her, only she has the insight that how the whole system work.

From above points, my feeling about the project is resonable.

> One blocking point today for its integration is the test. as you extend the API we need to test it and to ensure non regression in future. for this we need an application example that can run for [...]

I can add the new examples. But before I make any significant effort, I need the guarantee that the thing get the moving forward.

> Another concerns is the footprint penalty associated.
> We need to determine the impact of this additional code and decide if we should make it optional ( using compilation switch). But this point is minor...

arnopo, 2019-12-20:
> I can't find any link on this web that I can join the committee.
> Sorry, my word may hurt you, but it is my real feeling about OpenAMP community.

This year is a transition year for OpenAMP project. The governance has changed, the maintainers also.
I can understand your frustration because your patches are pending since a while, Next year we should come back in a normal flow. 

> I make a summary why we make all these change one month ago here:
> https://groups.google.com/forum/#!topic/open-amp/VEWXUbNF1No
> but nobody from the committee give any response.
>
>
> When will this year [...]
This is our priority from now. the release date in under discussion. 
> 
> And I guess that Wendy(the project leader) doesn't particent in this project anymore, is it right? I have many good technical conversation with her, only she has the insight that how the whole system work.

Yes Wendy is acting now on some other projects. I hope i also have a good insight of OpenAMP, as from this week ,  i am also OpenAMP maintainer with @edmooring. :-)
And your expertise is welcome in discussion , as you already do.

> From above points, my feeling about the project is resonable.
>
>
> I can add the new examples. But before I make any significant effort, I need the guarantee that the thing get the moving forward.

The fact that i'm now a maintainer and that i have already approve this patch should be a good guarantee. 

> 
> > Another concerns is the footprint penalty associated.
> > We need to determine the impact of this additional code and decide if we should make it optional ( using compilation switch). But this point is minor...

closed 2020-12-04 by arnopo