OpenAMP/open-amp #191: virtio_ring.h generates compiler warning

supergaute, 2020-02-13:
When compiling sources that includes open_amp.h and have the -Wcast-align warning enabled, I get this compiler warning:

```
In file included from ..../random_sourcefile.c:27:
..../openamp/virtio_ring.h: In function 'vring_init':
..../openamp/virtio_ring.h:130:13: warning: cast increases required alignment of target type [-Wcast-align]
  vr->desc = (struct vring_desc *)p;
             ^
..../openamp/virtio_ring.h:131:14: warning: cast increases required alignment of target type [-Wcast-align]
  vr->avail = (struct vring_avail *)(p + num * sizeof(struct vring_desc));
```

If you are not going to fix this, would it be ok to "hide" the vring_init() function in a C-file instead of having it inlined in the header file?

arnopo, 2020-02-13:
Thanks for reporting the issue!
Hiding this in the library does not seems  a good idea.  
Could you test following fix? this fixes the warnings on my side

```
@ -7,10 +7,12 @@
  */
 
 #ifndef VIRTIO_RING_H
 #define	VIRTIO_RING_H
 
+#include <openamp/compiler.h>
+
 #if defined __cplusplus
 extern "C" {
 #endif
 
 /* This marks a buffer as continuing via the next field. */
@@ -32,26 +34,28 @@ extern "C" {
 #define VRING_AVAIL_F_NO_INTERRUPT      1
 
 /* VirtIO ring descriptors: 16 bytes.
  * These can chain together via "next".
  */
+OPENAMP_PACKED_BEGIN
 struct vring_desc {
 	/* Address (guest-physical). */
 	uint64_t addr;
 	/* Length. */
 	uint32_t len;
 	/* The flags as indicated above. */
 	uint16_t flags;
 	/* We chain unused descriptors via this, too. */
 	uint16_t next;
-};
+}OPENAMP_PACKED_END;
 
+OPENAMP_PACKED_BEGIN
 struct vring_avail {
 	uint16_t flags;
 	uint16_t idx;
 	uint16_t ring[0];
-};
+}OPENAMP_PACKED_END;
 
 /* uint32_t is used here for ids for padding reasons. */
 struct vring_used_elem {
 	/* Index of start of used descriptor chain. */
 	uint32_t id;
```

supergaute, 2020-02-13:
I had to change `#include <openamp/compiler.h>` to`#include <metal/compiler.h>`, and `OPENAMP_PACKED_BEGIN` to `METAL_PACKED_BEGIN` and finally `OPENAMP_PACKED_END` to `METAL_PACKED_END` in order for this to compile.

After it builds it looks good, no warnings anymore.

supergaute, 2020-02-13:
Do you want me to make a pull request for this?

arnopo, 2020-02-13:
You should be able to use the  OPENAMP_PACKED_BEGIN...i would prefer to avoid a reference to libmetal here.
Please first figure out why you can not use OPENAMP_PACKED* . Then yes you can send a PR.
Thanks!

supergaute, 2020-02-13:
Here is what i did 10 seconds ago:
```
ogti@ubuntu:~/repo/open-amp$ git pull
Already up to date.
ogti@ubuntu:~/repo/open-amp$ git status 
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
ogti@ubuntu:~/repo/open-amp$ grep -r OPENAMP_PACKED_BEGIN
ogti@ubuntu:~/repo/open-amp$ 
```
Where is this macro supposed to be?

arnopo, 2020-02-14:
Oops, my apologize, you are right. I was pointing to a bad version during my test...
Ok to use METAL_PACKED

supergaute, 2020-02-14:
Ok, Done

arnopo, 2020-02-20:
fixed in https://github.com/OpenAMP/open-amp/pull/192

closed 2020-02-20 by arnopo