OpenAMP/open-amp #193: Realease 2020.compilation fails for linux/x86_64 machine

arnopo, 2020-02-17:
Regression found on example when compile on x86 machine.
```
/local/home/frq07267/views/openamp/open-amp/apps/system/linux/machine/generic/platform_info.c:26:10: fatal error: metal/scatterlist.h: No such file or directory
 #include <metal/scatterlist.h>
```
 Seems that a commit is missing in libmetal

arnopo, 2020-02-17:
@edmooring & @bentheredonethat
Please, could you croscheck and fix?

To reproduce issue on a Linux machine
1) Libmetal compilation 
 cmake . -Bbuild
cd build
make

2) openamp lib generation
cmake . -DWITH_APPS=on -DWITH_PROXY=off -Bbuild -DCMAKE_INCLUDE_PATH="../libmetal/build/lib/include" -DCMAKE_LIBRARY_PATH="../libmetal/build/lib"
cd build
make install DESTDIR=./bin
Thank,
Arnaud

arnopo, 2020-02-17:
Something is wrong here also:
https://github.com/OpenAMP/open-amp/blob/007f2fe640b9c2258aa4e6a1b038f51545dd49d8/apps/system/linux/machine/generic/platform_info.c#L243
This does not match with the libmetal API.

edmooring, 2020-02-19:
Yep. A stray local change crept in.

Just submitted a pull request to fix it.

Regards,
Ed M

arnopo, 2020-02-20:
fixed by https://github.com/OpenAMP/open-amp/pull/196 and https://github.com/OpenAMP/open-amp/pull/197

closed 2020-02-20 by arnopo