OpenAMP/open-amp #194: Adjust RPMSG_BUFFER_SIZE from build

supergaute, 2020-02-17:
Here it seems you allow the user to configure the RPMSG_BUFFER_SIZE:
https://github.com/OpenAMP/open-amp/blob/007f2fe640b9c2258aa4e6a1b038f51545dd49d8/lib/include/openamp/rpmsg_virtio.h#L24-L27
This define is used in rpmsg_virtio.c.
How can I configure this during the build process? It seems like should be a CMake option?

arnopo, 2020-02-17:
yes it possible to configure the RPMSG buffer size. 
But don't custom it if you use Linux Kernel for the main processor management . The RPMSG buffer size is hardcoded to 512 in the Linux Kernel!

To change the size you have to declare in cmake configuration file:
` add_definitions( -DRPMSG_BUFFER_SIZE=${RPMSG_BUFFER_SIZE} )`

An alternative to custome it using cmake command line,  is to add following line in the options.cmake:
```
if (DEFINED RPMSG_BUFFER_SIZE)
 add_definitions( -DRPMSG_BUFFER_SIZE=${RPMSG_BUFFER_SIZE} )
endif(DEFINED RPMSG_BUFFER_SIZE)
```
Then to generate the libray add `-DRPMSG_BUFFER_SIZE=xxxx` in your cmake command line

supergaute, 2020-02-17:
Yes, you would have to modify the cmake files in order to fix this to be possible. I will make a pull request...

closed 2020-02-17 by supergaute