OpenAMP/open-amp #210: Openamp on 2019.2 - fail to work with rpu-1

davidzehav1, 2020-05-17:
I have problem trying to work with openamp + share memory with r0 and r1
I want to work with openamp from kernel space and to have share memory for r0 and r1 - i have no problem (maybee better) to have 1 share memory for both r0 and r1 .
I was able to work and create channel for r0 - echo_test works fine but not for r1 (not getting the following message: virtio_rpmsg_bus virtio0: creating channel rpmsg-openamp-demo-channel addr 0x0 which i get only for r0)
Could you please help with this issue - can you share a device tree that will work with r0 and r1 (from kernel space) and what changes are needed (if any) in r1 for echo test application?
i tried to use the examples in the following links but they are not working:
https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/150568961/OpenAMP+2019.2
https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/118358017/OpenAMP+2019.1​
I am working with ZCU-102, vitis 2019.2 and petalinux 2019.2
Thanks
David,

arnopo, 2020-05-20:
@edmooring, @bentheredonethat,
Seems here related to Xilinx platform, could you have a look or give to @davidzehav1 an entry point for Xilinx support? 

Thanks,
Arnaud

bentheredonethat, 2020-06-07:
Hi David,

I am sorry for the delayed response. To be clear, is this linux on APU to RPU or RPU to RPU?

If with APU Linux can you provide device-tree? Also the firmware configuration information?  E.g. IPIs, Shared memory carveouts used for IPC.

Best Regards,
Ben

davidzehav1, 2020-06-08:
Hi Ben,

Thank you very much for your reply.

It's okay, I was manage to solve the issue - it was related to small change that I forgot to do on the RPU-1 FW (IPI Mask) and the way to operate echo_test for RPU-1 - about the last issue email was sent to Xilinx and it was promised that it will be fix soon

Thanks

David,

This email might have been sanitized in order to remove any potentially malicious embedded objects.

This message (including any attachments) issued by RAFAEL- ADVANCED DEFENSE SYSTEMS LTD. (hereinafter "RAFAEL") contains confidential information intended for a specific individual and purpose, may constitute information that is privileged or confidential or otherwise protected from disclosure. If you are not the intended recipient, you should contact us immediately and thereafter delete this message from your system. You are hereby notified that any disclosure, copying, dissemination, distribution or forwarding of this message, or the taking of any action based on it, is strictly prohibited. If you have received this e-mail in error, please notify us immediately by e-mail mailto:lawraf@rafael.co.il and completely delete or destroy any and all electronic or other copies of the original message and any attachments thereof.

arnopo, 2020-06-23:
Seems that the issue can be closed, please reopen it if it is not the case

closed 2020-06-23 by arnopo