OpenAMP/open-amp #211: fail to compile open-amp master branch

SlevinZhang, 2020-05-19:
I try to compile the open-amp on a ubuntu machine. And I ran into some undefined reference error.
**CMake Config**
```bash
cmake ../ -DCMAKE_INCLUDE_PATH=../../libmetal/install-libmetal/usr/local/include -DCMAKE_LIBRARY_PATH=../../libmetal/install-libmetal/usr/local/lib -DWITH_APPS=ON -DLIBMETAL_LIB=../../libmetal/install-libmetal/usr/local/lib/ -DLIBMETAL_INCLUDE_DIR=../../libmetal/install-libmetal/usr/local/include
```
PS: I have to add LIBMETAL_LIB and LIBMETAL_INCLUDE_DIR, otherwise i will receive some error when I config cmake.

**Make CLI**
```bash
make VERBOSE=1 DESTDIR=../install-libmetal/ install
```
**Comments**
I think I build libmetal library successfully and I could run the libmetal demo applicaiton. 

**Error Message**
[ 18%] Linking C executable msg-test-rpmsg-ping-static
cd /home/siliang/Workspace/Envision/system_software/open-amp/build/apps/tests/msg && /usr/bin/cmake -E cmake_link_script CMakeFiles/msg-test-rpmsg-ping-static.dir/link.txt --verbose=1
/usr/bin/cc  -Wall -Wextra  -rdynamic CMakeFiles/msg-test-rpmsg-ping-static.dir/__/__/system/linux/machine/generic/helper.c.o CMakeFiles/msg-test-rpmsg-ping-static.dir/__/__/system/linux/machine/generic/rsc_table.c.o CMakeFiles/msg-test-rpmsg-ping-static.dir/__/__/system/linux/machine/generic/platform_info.c.o CMakeFiles/msg-test-rpmsg-ping-static.dir/rpmsg-ping.c.o  -o msg-test-rpmsg-ping-static  -L/home/siliang/Workspace/Envision/system_software/libmetal/install-libmetal/usr/local  -L/home/siliang/Workspace/Envision/system_software/open-amp/build/lib -Wl,-rpath,/home/siliang/Workspace/Envision/system_software/libmetal/install-libmetal/usr/local:/home/siliang/Workspace/Envision/system_software/open-amp/build/lib: ../../../lib/libopen_amp.a 
CMakeFiles/msg-test-rpmsg-ping-static.dir/__/__/system/linux/machine/generic/helper.c.o: In function `init_system':
helper.c:(.text+0x1a): undefined reference to `metal_default_log_handler'
helper.c:(.text+0x38): undefined reference to `metal_init'

arnopo, 2020-05-19:
hi @SlevinZhang,
Here is a simple bash script to test openamp on an Ubuntu PC
This script has to be run in the directory containing the libmetal and the open-amp repos. Perhaps you will have to adapt it to your environment...

```
#!/bin/bash
 
dirs_to_check=$PWD
echo "################### build libmetal #####################"
cd libmetal
rm -r build
cmake . -Bbuild || exit 1
cd build
make || exit 1

cd $dirs_to_check

echo "################### build openamp #####################"
cd open-amp
rm -r build
cmake . -DWITH_APPS=on -DWITH_PROXY=on -Bbuild -DCMAKE_INCLUDE_PATH="../libmetal/build/lib/include" -DCMAKE_LIBRARY_PATH="../libmetal/build/lib" || exit 1
cd build
make install DESTDIR=./bin || exit 1

cd $dirs_to_check
cd open-amp
cd build
echo "################### run ping test #####################"
LD_LIBRARY_PATH=./bin/usr/local/lib:../../libmetal/build/lib/  ./bin/usr/local/bin/rpmsg-echo-shared &
sleep 1
LD_LIBRARY_PATH=./bin/usr/local/lib:../../libmetal/build/lib/  ./bin/usr/local/bin/msg-test-rpmsg-ping-static 1 || exit
```
hope that this will help

SlevinZhang, 2020-05-19:
Awesome, thank you for your help.

arnopo, 2020-05-25:
@SlevinZhang 
So i consider this issue as closed, don't hesitate to reopen it if it is not the case.

closed 2020-05-25 by arnopo