OpenAMP/open-amp #217: thread-safe

soso7885, 2020-08-06:
Hi All

is openAMP support thread-safe ? now I'm facing the deadlock issue, 
the remote processor (arm-cm4 with rtos) create 5 ept, and the main processor (arm-ca73 with linux kernel) also create 5 ept,
stress testing the data transmit (each ept send the data to otherside, then send back).
the rtos side (cm4) hanging in rpmsg_virtio_rx_callback randomly
`
rpmsg_virtio_rx_callback
{
   ....
    metal_mutex_acquire(&rdev->lock);  <--- hang here, cannot get the lock
    rp_hdr = rpmsg_virtio_get_rx_buffer(rvdev, &len, &idx);
    metal_mutex_release(&rdev->lock);
 ....
}
`

arnopo, 2020-08-24:
@soso7885 
Please could you detail your environment? 
- Which OS do you use on the CM4?
- Is the rpmsg_virtio_rx_callback is called under interrupt context?

The mutex is shared between the RX and The TX. Treating the Rx message under interrupt context while Tx message are sent in normal context leads to this kind of lock.

jorisvanderpol, 2020-11-25:
Hi,

I am interested in this subject as well. Is it possible to poll for new rpmsg virtio messages from one thread and send rpmsg virtio messages from another? I am using the Linux user space OpenAMP drivers. I was under the assumption that it is possible, due to the two vrings.

Kind regards,

Joris van der Pol

arnopo, 2020-11-26:
Hi @jorisvanderpol 
There is mutexes mechanism in the library to protect from concurrent access. For Linux, this relies on the following code
https://github.com/OpenAMP/libmetal/blob/master/lib/system/linux/mutex.h
Personally, I have not implemented OpenAMP for the Linux user space, but as long as threads can be preempted, I don't seeany problem to implement receiving and transmitting in separate threads.

arnopo, 2020-11-26:
> Hi All
>
> is openAMP support thread-safe ? now I'm facing the deadlock issue,
> the remote processor (arm-cm4 with rtos) create 5 ept, and the main processor (arm-ca73 with linux kernel) also create 5 [...]

@soso7885 
Any news about your issue?

supergaute, 2020-11-27:
Could this be related to related to OpenAMP/libmetal#101?

huegelcesar, 2022-01-07:
Hi folks,

I'm facing the same issue. A deadlock situation comes hand when calling `rpmsg_virtio_rx_callback` and/or `rpmsg_virtio_send_offchannel_raw` (`rpmsg_virtio.c`), although it's not being called from interrupt context.

Just to give you a bit more context, I have a lower priority task, which calls `platform_poll` and a higher priority task which is woken by another interruption (DMA), in a high frequency, and then sends a message through the RPMSG channel.

I have read OpenAMP/libmetal#101 and applied https://github.com/OpenAMP/libmetal/pull/148 but still struggling to avoid this deadlock.

@soso7885 I'm not sure it's the exactly same scenario as yours, but did you get this problem solved?

Thank you all in advance!

arnopo, 2023-03-15:
Close it as https://github.com/OpenAMP/libmetal/issues/101 is now closed

closed 2023-03-15 by arnopo