OpenAMP/open-amp #218: Firmware load example on Linux

jorisvanderpol, 2020-09-21:
Hi,

I am working on a project that requires the firmware to be loaded using remoteproc on a Linux ARM64 target. This there an example available, or will there be? I having difficulty building an app using the image_store_ops methods. The firmware is already built using Xilinx Vitis, and works when loading using the FSBL.

Kind regards,

Joris

edmooring, 2020-09-22:
Hi,

What platform are you using? If you’re running Linux on ARM64 and loading firmware onto a remote processor it can be as simple as:

1. Put the firmware in /lib/firmware
2. Load the appropriate remoteproc driver, if it isn’t already loaded.
3. # echo $firmware-binary-name > /sys/class/remoteproc/remoteproc0/firmware
4. # echo start > /sys/class/remoteproc/remoteproc0/state

Regards,
Ed M

edmooring, 2020-09-22:
Hi,

I just remembered that you do need to update your device tree appropriately. If you include the open-amp package group in your Linux build, it should be added. If not, there are instructions in UG1186.

Regards,
Ed M

jorisvanderpol, 2020-09-24:
Hi Ed,

I am using the Xilinx ZynqMP UltraScale+. PetaLinux runs on the application cores. The firmware for the real-time cores must be loaded by the application. Also, PetaLinux is built by another party, so the goal is to keep the interfaces with the OS (e.g. device tree) as small as possible. The idea is to place the firmware into the application that communicates with that specific firmware and load it from memory to the physical memory that is reserved for the real-time core. The only thing needed is to place a reserve memory node in the device tree and some IPI stuff.

The communication between the application and firmware counter-part works fine when either using the FSBL or the way suggested by you.  So the next step is to let the application load the firmware from memory. But loading from memory seems to fail. And I haven't figure out yet how to start the firmware via remoteproc. I looked at the bare metal firmware load example, so I was wondering if there is a Linux variant.

Thanks and regards,

Joris

edmooring, 2020-09-25:
Hi Joris,

I don't think that what you want to do is actually doable. The problem is that configuring the RPU when Linux is running on the APU has to be done through the kernel via EEMI calls. There are no user-space interfaces to those calls available. The only way to start and stop the RPUs and configure their associated TCM from user space is via remoteproc and its syses interface.

jorisvanderpol, 2020-09-28:
Hi Ed,

I understand, hence Xilinx' ZynqMP R5 remoteproc driver. The example in UG1186 uses kernel drivers for life cycle management and inter-processor communication. The Xilinx website offers a hybrid, LCM via the kernel driver and IPC via user-space drivers. As far as I understand, the data transfer is faster via user-space drivers and a crash in these drivers do not affect the kernel. Is this correct? Are there any other benefits or drawbacks?

Thanks for the help so far an regards,

Joris

edmooring, 2020-10-03:
Hi Joris,
The IPC via user-space drivers approach is used a lot internally at Xilinx for experimentation. It can be faster, and it's definitely more flexible. There aren't any major downsides to the user-space approach.

Is it OK if I close this issue, since I think you have the answer to your question?

jorisvanderpol, 2020-10-05:
Hi Ed,

Yes, and thanks for the help!

closed 2020-10-05 by arnopo