OpenAMP/open-amp #220: rpmsg: supress broken  rpmsg_init_ept API

arnopo, 2020-10-05:
The rpmsg_init_ept does not seem to be usable by application but is used internally. The proposal is to mark this API deprecated  API and in at least two full release, to remove this API as the standard API to use is  rpmsg_create_ept.
In parallel, the online function must be moved to rpmsg_internal.h

Issue entered in reference to :

- https://github.com/OpenAMP/open-amp/pull/204#discussion_r423218842

- https://github.com/OpenAMP/open-amp/pull/208#pullrequestreview-420375651

arnopo, 2020-10-05:
Hi @Ed,
Proposal is to use similar mechanism as one used in zephyr (https://docs.zephyrproject.org/latest/development_process/api_lifecycle.html)
Introducing the deprecated attribute ( supported by gcc and IAR) allows to continue to use the API, but with a warning at compilation time:
` warning: ‘rpmsg_init_ept’ is deprecated [-Wdeprecated-declarations]`

To deprecate a function we just need to update the function declaration in the API header:
```
-static inline void rpmsg_init_ept(struct rpmsg_endpoint *ept,
+__deprecated static inline void rpmsg_init_ept(struct rpmsg_endpoint *ept,

```
and the "-Wno-deprecated-declarations" compilation option can disable the warning on need.

What is your feeling about this solution?
i plan to send associated PR tomorrow if this looks good to you.

edmooring, 2020-10-05:
Hi @arnopo,
This seems to me to be a good solution.

closed 2020-10-12 by arnopo