OpenAMP/open-amp #235: Copyright and license in apps/examples

takayuki-imada, 2020-10-26:
Could you please add copyright and license information to source files in apps/examples/{echo,matrix_multiply,rpc_demo,rpmsg_sample_echo}/ ?

The source files don't have any information about them.

Thanks,

arnopo, 2020-11-05:
@takayuki-imada 
Thanks for the information!  Is the copyright also an issue for you or only the licensing?
Feel free to propose a fix to accelerate the correction.

takayuki-imada, 2020-11-10:
@arnopo 
Licensing is my concern.
I just want to know what kind of license is applied to the source files I pointed out.
BSD-3(or2)-Clause as shown in LICENSE.md, or another?

arnopo, 2020-11-10:
Yes  as mentioned in https://github.com/OpenAMP/open-amp/blob/master/LICENSE.md, you can use the BSD-3-Clause

takayuki-imada, 2020-11-10:
Thanks, that's good enough information to me!

arnopo, 2020-11-10:
Reopen as we need to fix it for next release

arnopo, 2020-11-24:
issue addressed by https://github.com/OpenAMP/open-amp/pull/245

closed 2020-12-01 by arnopo