OpenAMP/open-amp #236: remoteproc_mmap returns address which can be 0

bentheredonethat, 2020-10-26:
1 solution is to move returned address to be an argument and make return value success/failure-reason

arnopo, 2020-11-02:
Do you have a use case that highlights the fact that 0 can be a valid virtual address?
If 0 can be valid, yes, this API needs to be updated, if not, returning NULL in case of failure seems reasonable to me.

bentheredonethat, 2020-11-02:
yes address can be 0 on Xilinx Zynq UltraScale Plus on RPU running baremetal or FreeRTOS application. The address 0 relates to the lowest value on TCM bank 0A for RPU0 and TCM bank 1A for RPU1.

arnopo, 2020-11-05:
what about returning METAL_BAD_VA?

edmooring, 2020-11-06:
That is probably the best solution, since it doesn't change the documented API. However, the comments in openamp/remoteproc.h need to be enhanced to show the error return value, and any examples that depend on the previous behavior should also be updated. It should probably also be highlighted in the release notes.