OpenAMP/open-amp #237: Ping application hang after restart

jorisvanderpol, 2020-10-27:
Hi,

I noticed that the ping application of the echo example hangs after starting it again (twice in a row). I am running Linux on the APUs and bare-metal on one of the RPUs (Xilinx UltraScale+ MPSoC, PetaLinux 2019.2).

Kind regards,

Joris van der Pol

bentheredonethat, 2020-10-27:
Hi Joris,

Thanks for the issue. You say "starting it again". is this system wide? Do you restart the whole system or just APU? also is the echo example for using the OpenAMP library and RPMsg in linux userspace or via the Xilinx Remoteproc R5 Kernel driver?

Thanks
Ben

jorisvanderpol, 2020-10-27:
Hi Ben,

I restart the ping application only when it is finished. The system is not restarted, that is, Linux keeps running on the APU and nothing is done with respect to the RPU. I use the user space drivers.

The RPU application is added to the boot image file of PetaLinux, so it will be booted when the system is booted.

Kind regards,

Joris

bentheredonethat, 2020-10-27:
Understood. In this case the RPU program will finish running after the first time. you can specify the number of runs at the command line but otherwise you will have to unload and load new firmware to run the application again. 

If you are looking for an example of a program running a loop waiting to only finish explicitly at command of user please check the proxy (rpc) demo application.

If that is not sufficient, possibly adding more detail on your use case might help

Hope this helps
Ben

jorisvanderpol, 2020-10-28:
Hi Ben,

I'll check the rpc demo application. The use case is as follows: the RPU (slave) application is booted at system startup. It will just wait until the APU (master) application comes up, communication is setup between the two and the RPU application handles all requests from the RPU application.  However, the APU application can be restarted, meaning the RPU application must not exit but be ready to handle the communication setup again and handle more requests. Is this even possible?

Kind regards and thanks for the help,

Joris

bentheredonethat, 2020-10-28:
the proxy app will still require additional modification as it expects the endpoint to be up. In this case you will require the APU and RPU to re-establish endpoints or dynamic rediscovery of endpoints.

jorisvanderpol, 2020-11-02:
I will close this as it is not an issue and thanks for the help.

Joris

closed 2020-11-02 by jorisvanderpol

Fei-Ho, 2024-03-22:
I also encountered this problem, can you tell me how to implement re-establish endpoints or dynamic rediscovery of endpoints?

Thank you so much !