OpenAMP/open-amp #242: Can I use openAMP with linux Master and slave scenario?

arsieroluca, 2020-11-07:
I found are possible many combinations of use for openAMP (linux and baremetal, linux and rtos, baremetal and baremetal ...) and among these it seems possible to have a inux master and a linux slave. 

It is possible to run them together and exchange data using openamp?

Any idea about this?

arnopo, 2020-11-09:
Hi @arsieroluca,
Could you details your usecase  beccaus I'm not sure to understand your need...
RPMsg protocol is bidirectional, so one instance ( slave or master) should be sufficient to  exchange data with  a remote processor

arsieroluca, 2020-11-09:
Hi @arnopo ,

I was referring to such a scenario:

| rpMsg App   |     <====>  | rpMsg App |
| Linux Master |                     | Linux Slave  |

But I don't find any example...

Thank you...

arnopo, 2020-11-09:
Here you can find examples:
https://github.com/OpenAMP/open-amp/tree/master/apps/examples

For instance:
https://github.com/OpenAMP/open-amp/blob/master/apps/examples/echo/rpmsg-ping.c implement the master
https://github.com/OpenAMP/open-amp/blob/master/apps/examples/echo/rpmsg-echo.c  implement the slave

If you want to play this example on your Linux PC , here a script that you can run ( could need to be adapted to you environment).

```
echo "################### get libraries #####################"
git clone https://github.com/OpenAMP/open-amp.git
git clone https://github.com/OpenAMP/libmetal.git

dirs_to_check=$PWD
echo "################### build libmetal #####################"
cd libmetal
rm -r build
cmake . -Bbuild || exit 1
cd build
make || exit 1

cd $dirs_to_check

echo "################### build openamp #####################"
cd open-amp
rm -r build
cmake . -DWITH_APPS=on -DWITH_PROXY=on -Bbuild -DCMAKE_INCLUDE_PATH="../libmetal/build/lib/include" -DCMAKE_LIBRARY_PATH="../libmetal/build/lib" || exit 1
cd build
make install DESTDIR=./bin || exit 1

cd $dirs_to_check
cd open-amp
cd build
echo "################### run ping test #####################"
LD_LIBRARY_PATH=./bin/usr/local/lib:../../libmetal/build/lib/  ./bin/usr/local/bin/rpmsg-echo-shared &
sleep 1
LD_LIBRARY_PATH=./bin/usr/local/lib:../../libmetal/build/lib/  ./bin/usr/local/bin/msg-test-rpmsg-ping-static 1 || exit

```

arsieroluca, 2020-11-09:
Hi @arnopo ,

thank you for your suggestions... but if I'm right this is Linux Master <---> baremetal

I need linux Master <---> linux slave

both linux have to do different stuff but use rpMsg ..

arnopo, 2020-11-09:
> Hi @arnopo ,
> 
> thank you for your suggestions... but if I'm right this is Linux Master <---> baremetal
No, this implements Linux<->Linux. Two Linux applications communicate together through a socket. one application is slave the other is master. The difference in this mode is that it is running on the same Linux so  the message notification is relying on a socket instead of a mailbox or something similar.
that's means that the main difference with your use case is the implementation of the [platform_info](https://github.com/OpenAMP/open-amp/blob/master/apps/system/linux/machine/generic/platform_info.c) code. You will have to adapt it on your platform to correctly define the shared memory and notification mechanism. 

> 
> I need linux Master <---> linux slave
> 
> both linux have to do different stuff but use rpMsg ..

For Linux you have 2 ways to use RPMsg:
* use the RPMsg framework implemented in Linux kernel (master only)
* use OpenAMP library in user space that supports both master and slave. The example I pointed out is based on that, except that the link is done between two binaries running on the same Linux.

If this is not enough, @edmooring may have other examples?  But you will probably have to do the same thing, adapt the code to your own platform.

arsieroluca, 2020-11-09:
Hi @arnopo,

thank you for your suggestions.

What I have in mind is to have two linux applications communicate on two different linuxes.

One with some peripherals on core 0 and the other with other peripherals on core 1 and compare the data processed in the two applications and on the two linux.

> use OpenAMP library in user space that supports both master and slave. The example I pointed out is based on that, except that the link is done between two binaries running on the same Linux.

almost... but on the different linux running each on different core..

- is there a way to do this?

Thank's a lot

arnopo, 2020-11-13:
> * is there a way to do this?
this part depends on your platform,  you will have to declare the shared memories for RPMsg communication and optionally implement a "doorbell" mechanism to kick the other core that a new message is available.
Have a look to https://github.com/OpenAMP/open-amp/tree/master/apps/machine example that this can help to understand how you can implement this.

arsieroluca, 2020-11-13:
Thank you @arnopo !
The way that you suggest is what I searching for...

I thought about a shared memory mechanism...and I understand openamp give me an help in that...

Zynq platform is a good starting point for me...
- To use two shared memory between two linux, if I'm right, I need two dtb (dts file...) and a bootloader changed to load kernel and dtb on each core, is right?

Thank you for the example

edmooring, 2020-11-14:
> Thank you @arnopo !
> The way that you suggest is what I searching for...
>
> I thought about a shared memory mechanism...and I understand openamp give me an help in that...
>
> Zynq platform is a good [...]

This is a lot more difficult than it looks. The problem is that with 2 instances of Linux, each is going to want to control all the resources of the system. Some resources, such as memory, can be partitioned using the device tree. Unfortunately, there are other devices, like interrupt controllers and control registers, that cannot. As far as I can tell, Xilinx does not support two instances of Linux running on the Zynq.

If you want to experiment with a configuration with 2 or more instances of Linux, I would suggest using a hypervisor like Xen, and assigning certain peripherals to each Linux instance via passthrough, and setting up an appropriate shared memory region for OpenAMP communication. This should be much easier, but it will still have its challenges.

arsieroluca, 2020-11-14:
@edmooring and @arnopo now I have two way ;)!

In my research before asking I read about XEN but I don't understand if it is too onerous for a Zynq or for a x86 dual core (these are the platform on would to try).

- I guess the peripheral configuration problem is common however ... I think the interrupt controller should still be used by both cores, right?

- Do I need to change the device tree properly or need other kernel-level changes?

- Is there any example that allows to get two linuxes with Xen and openAMP?

Thank's a lot

edmooring, 2020-12-16:
Sorry for the delay in getting back to you.
I'm not sure the Zynq has quite enough power to support running Xen. Pretty much any X86 box from the last 10 years can handle it just fine. 

- The interrupt controller can only be used by one Linux instance. The other Linux will have to be modified to not use it, or both Linux instances can run under control of Xen.
- There will be device tree changes required.
- I don't know of an example of domU to domU communication via Xen and OpenAMP.

This should be a really good opportunity for learning :) should you decide to move forward with it.

closed 2021-04-28 by arnopo