OpenAMP/open-amp #249: "Is not a head" error thrown after a random time on multiple endpoint rpmsg transmission

saravanakumar-sethuraman, 2020-12-05:
The issue concerns the source file **`virtio_rpmsg_bus.c`** (drivers/rpmsg/) in the upstream Linux kernel that follows the reference OpenAmp implementation:-

https://github.com/OpenAMP/open-amp/blob/master/lib/rpmsg/rpmsg_virtio.c
https://github.com/OpenAMP/open-amp/blob/master/lib/virtio/virtqueue.c

**`Scenario`**:- RPMSG endpoints are exposed as TTY devices /dev/ttyRPMSG30, /dev/ttyRPMSG60. Three different processes write to these devices at various cyclic intervals (100ms, 20ms) small messages of sizes like 400, 60 bytes. After a randome time, A35 to M4 transmission stops with an error : 

virtio_rpmsg_bus virtio1: output:id **48** is not a head!
imx_rpmsg_tty virtio1.rpmsgM4A35.-1.30: virtqueue_add_outbuf failed: -5
imx_rpmsg_tty virtio1.rpmsgM4A35.-1.30: rpmsg_send failed: -5

Even after the issue occurs affecting A35 ==> M4 transmission, the other direction M4 ==> A35 transmission still continue to work fine. This shows that A35 Tx VRING has gone bad while the A35 Rx VRING continue to work without any issues.
In order to just give some background information on the debug steps, we have done, I list here certain debug logs we see:-
These logs are printed out of the **`virtqueue_add_outbuf()`** function:-

**Three observations:**-

**HEADs out of sequence** ==> This is perhaps normal when there are concurrent senders or caller of the function rpmsg_send_offchannel_raw(). HEADs that get dequeued from the USED ring need not necessarily get enqueued into the AVAIL ring in the same order, which explains that we see an out of sequence of the HEADs in the logs.
Added buffer head 88 to ffff80000c566000
Added buffer head 89 to ffff80000c566000
Added buffer head 91 to ffff80000c566000
Added buffer head 93 to ffff80000c566000
Added buffer head 90 to ffff80000c566000
Added buffer head 92 to ffff80000c566000
Added buffer head 95 to ffff80000c566000
Added buffer head 94 to ffff80000c566000

**HEAD missing and 3. Duplicated HEAD**
Added buffer head 14 to ffff80000c566000 |   |  
Added buffer head 15 to ffff80000c566000 |   |  
Added buffer head 17 to ffff80000c566000 |   |  
Added buffer head 18 to ffff80000c566000 |   |  
Added buffer head 21 to ffff80000c566000 | 19 | Missing
Added buffer head **20** to ffff80000c566000 | 20 | Duplicated
Added buffer head 22 to ffff80000c566000 | 21 | Out of Sequence
Added buffer head **20** to ffff80000c566000 | 22 | Out of Sequence

Despite the duplication of the Buffer Head, the transmission continues to work until we reach a stage when a duplicated HEAD remains in detached state (by a call to **`virtqueue_get_buf()`**) but not yet enqueued into the AVAIL ring but the process from which this specific transmission request originates gets pre-empted after the release of the mutex **`tx_lock`**, and further transmit requests call **`virtqueue_get_buf()`** retrieving a slot from the USED ring and when the retrieved slot is the duplicated HEAD that is still previously remaining in detached state because it was not yet added to the AVAIL ring by a call to **`virtqueue_add_outbuf()`**, does the issue "is not a head" occur.

So the key to finding a solution to the problem is to understand why do we get duplicated HEADs. If my understanding is correct, at any time the USED and AVAIL ring contains the Buffer Descriptor HEADs (with the IDX pointing to the one actually available or used) but all these HEADS need to be unique and there should not be any duplications. What happens over and over again is the buffers get recycled over muliple transmissions that keep happening.

Does anyone have any hints from the observations above ?

Thanks and Regards,
Saravana

arnopo, 2020-12-10:
Hi @saravanakumar-sethuraman,
Not sure that is the good place for your post, as issue seems not related to the OpenAMP library, or i missed something?

edmooring, 2020-12-17:
Can you try defining VQUEUE_DEBUG in your build for the M4? If there is a problem with re-entrancy on the M4 side, it's possible that enqueueing the buffer to the free list there could corrupt things. Enabling debugging might catch the issue.

arnopo, 2021-04-06:
Hi @saravanakumar-sethuraman,
There does not seem to be any activity on this issue. 
Have you found the issue? Can we close the point?

closed 2021-04-28 by arnopo

gmth-penko, 2025-01-28:
Commenting to help others finding this topic, it is the only Google hit matching:

We had the same issue, in our case it was caused by a double `rpmsg_queue_nocopy_free(rx_buf)` on the embedded processor side (m4). Not an OpenAMP issue, not a Linux issue.