OpenAMP/open-amp #250: Add semantic versioning

jorisvanderpol, 2020-12-09:
Please add semantic versioning to OpenAMP so it can be traced and compared when running on different cores.

arnopo, 2020-12-10:
Please could you details your exact need? 
Do you need the existing version of the library to be updated at the time of release or do you need another way to get the version?

jorisvanderpol, 2020-12-15:
Hi Arno,

The same as the metal library actually ([https://github.com/OpenAMP/libmetal/blob/master/lib/version.h](https://github.com/OpenAMP/libmetal/blob/master/lib/version.h)). A major, minor and patch version to track the changes over time.

I use a Xilinx ZCU106 board, and have setup a communication channel between between the APU running Linux and 1 RPU core running a bare-metal application. The OpenAMP library on the APU side comes from the PetaLinux SDK and on the RPU side from the ZCU106 default BSP. I noticed that in the past there were minor differences when browsing the library code in the SDK compared to the BSP. For problem solving it would be easy to trace a version (which the library user can get from the major, minor, patch integers) It should become apparent directly that the 2 differ. That does not mean it is an issue in itself, but it gives an indication.

I would be expect when a new implementation is released, a version is assigned that represents interface changes, implementation changes only and bug fixes. I trace these versions in my code for libmetal already although I think the libmetal version has not been updated recently.

Kind regards,

Joris

edmooring, 2021-01-11:
Hi Joris,
This is a reasonable idea, but won't solve your current problems. Unfortunately, the APU code and RPU BSP code come from different repositories with different maintainers within Xilinx. The code bases have diverged somewhat, and both are different from the OpenAMP Project repos. 

That being said, I would support a pull request to add this to OpenAMP.

Regards,
Ed M

closed 2021-04-26 by arnopo