OpenAMP/open-amp #251: App: add an option to build TESTS and EXAMPLES code

vishnu-banavath, 2020-12-10:
Hi, 
This is more of a question or suggestion. I was wondering an option be provided to the user to build TESTS and EXAMPLES instead of building it by default when WITH_APPS defined. Another way of looking at is, most of the platform code is under APPS directory. Is there any chance to add a PLATFORMS directory under openamp root directory and move all the platform specific code under this directory?

Following is check code I was referring in my statement above

diff --git a/apps/CMakeLists.txt b/apps/CMakeLists.txt
index 95b48e0..c6297fa 100644
--- a/apps/CMakeLists.txt
+++ b/apps/CMakeLists.txt
@@ -10,5 +10,11 @@ set (APPS_SHARE_DIR "${CMAKE_CURRENT_BINARY_DIR}/share")

 add_subdirectory (machine)
 add_subdirectory (system)
+
+if(WITH_TESTS)
 add_subdirectory (tests)
+endif(WITH_TESTS)
+
+if(WITH_EXAMPLES)
 add_subdirectory (examples)
+endif(WITH_EXAMPLES)

Thanks

arnopo, 2020-12-10:
Hi @vishnu-banavath,
Thanks for the suggestion! 
Personally i don't see a concrete interest of splitting the apps in 2 with build options. 
Have you something in mind behind this proposal?
Moreover there is a dependency between tests and examples. For instance the ping test need the echo example.

vishnu-banavath, 2020-12-10:
Hi @arnopo, thanks for your quick reply.
I personally think platforms code should be outside of app directory. As an example, platform code can be a shared library which can be linked against different applications. I don't know how it might work in terms of baremetal and RTOS. 
Tests and examples code can still live under apps and can have inter-dependency.

This is just my thought, but if you think it does not add much value, feel free to close this thread.

Thanks

arnopo, 2020-12-15:
Today the platform code is adapted for the examples and tests. But having something more generic for ease the port of the openAMP is something interesting to have.
In any case this would probably remain an example as this depend on the OS, the platform but also the project(e.g. the number of RPMsg buffer, the resource table content...)

xiaoxiang781216, 2020-12-15:
If we want to implement some generic service(e.g. tty, log, fs), it's important to decouple the platform code from the service code.

closed 2023-05-09 by arnopo