OpenAMP/open-amp #269: Multiple endpoints but one interrupt?

infn-ke, 2021-02-17:
Does rpmsg support multiple endpoints that all use the same interrupt?

arnopo, 2021-02-18:
I'm not sure to understand your question...so please clarify your question if my answer is not enough accurate.
When you create an endpoint you provide a callback. This callback is called by the RPMsg framework. so it depends in which context you call rproc_virtio_notified to notify the RPMsg framework that new messages are available. 
If the rproc_virtio_notified is called under interrupt context your ept callbacks are also called under interrupt context... Not recommended...

bentheredonethat, 2021-02-24:
Hi @infn-ke  this might be hardware specific as well. On Zynq UltraScale Plus and Versal SoC's this is possible in theory.

As the IPIs on both SoC's mentioned have multiple apertures in their respective IPI message buffers. 
see https://www.xilinx.com/support/documentation/user_guides/ug1085-zynq-ultrascale-trm.pdf IPI message buffers section.

infn-ke, 2021-02-26:
I want to understand how openamp solves multiple endpoints and concurrency. E.g. say I'm running two Linux processes communicating over two different endpoints. Can that be achieved with one interrupt? Will UIO deliver that interrupt to both processes and how is the arbitration to the shared memory solved (process semaphores?)

closed 2021-03-10 by arnopo