OpenAMP/open-amp #270: rpmsg without remoteproc?

infn-ke, 2021-02-26:
Can I use rpmsg without remoteproc or are they coupled? I only want to use rpmsg to communicate over a shared memory. I have two UIO interfaces, one for the interrupt and one for the shared memory, what is the required steps to do that?

arnopo, 2021-02-26:
Please have look to this Zephyr sample: https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/subsys/ipc/openamp
it implements the RPMsg without remoteproc.
Another interesting link that implements an abstraction layer:
https://github.com/zephyrproject-rtos/zephyr/tree/master/subsys/ipc/rpmsg_service

infn-ke, 2021-02-26:
> Please have look to this Zephyr sample: https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/subsys/ipc/openamp
> it implements the RPMsg without remoteproc.
> Another interesting link that [...]

Thanks will have a look at it.

infn-ke, 2021-03-02:
@arnopo the Zephyr doc states that one of the apps can run with Linux rpmsg_client_sample. But how to provide shared memory and interrupt resources to the rpmsg_client_sample kernel module?

arnopo, 2021-03-03:
The Zephyr application you mention is :
https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/subsys/ipc/openamp_rsc_table
This sample implements only the slave part, relying on the resource table, so the OpenAMP remoteproc.

rpmsg_client_sample is a Linux kernel sample relying to the rpmsg, remoteproc and mailbox Linux kernel frameworks ( not use the OpenAMP lib). The  IRQ are managed by the mailbox Linux kernel framework, the  shared memory is managed the remoteproc Linux kernel framework.
So this example does probably not match with your expectation.

Regarding your initial request (based on OpenAMP lib): the IRQs and shared memories are generally implemented in a back-end that is specific to your platform. There is not defined steps but some examples. 
Zephyr provides some, the OpenAMP provides some other (but based on the remoteproc) implemented in the platform_info.c files.  
I therefore suggest that you take one of these examples as a starting point and adapt it to your platform need.

infn-ke, 2021-03-04:
But is the zephyr sample compatible with Linux kernel implementation of rpmsg? E.g. `VRING_TX_ADDRESS` is placed somewhere in the shared memory (`VDEV_START_ADDR + SHM_SIZE`). I thought they ring would start at `SHM_START_ADDR`?

https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/subsys/ipc/openamp/common.h#L23