OpenAMP/open-amp #292: VirtQueue should use metal_cache_* APIs

carlocaione, 2021-06-03:
On some platforms the VirtQueue and Vring structures could be stored in cacheable memory. When a data cache is present OpenAPM should take care of flushing / invalidating vring structures and control block when needed to keep consistency, this should be done using the `metal_cache_*` APIs.

arnopo, 2021-06-03:
As the cache management is platform dependent, the libmetal is a better place to handle the cache. 
The open-amp lib should not have to care about the cache memory management.
Regarding your patch in libmetal, I suppose that your need is for Zephyr, right?
Here is an example in libmetal for NuttX than could be adapted for Zephyr:
https://github.com/OpenAMP/libmetal/blob/master/lib/system/nuttx/io.c

carlocaione, 2021-06-03:
@arnopo agree but unfortunately I don't think that's enough. When also the vring control block is in cacheable memory (and shared between remote and master) we need to flush / invalidate also when accessing the struct that is in the shared memory, for example https://github.com/zephyrproject-rtos/open-amp/blob/master/open-amp/lib/virtio/virtqueue.c#L550

arnopo, 2021-06-07:
You are right! Thanks for pointing  this out to me.

That would means that we should have to use io access to update the virtqueues or to introduce a cmake option for the cache management.

Before we get into the details. It seems to me that you are the first one to point this out and ask for the use of cache memory for RPMsg communication.
Could you explain the rationale behind this request?
Do you observe some performance improvement using cacheable memory?

carlocaione, 2021-06-08:
> Could you explain the rationale behind this request?
> Do you observe some performance improvement using cacheable memory?

I'm not sure how much I can go into details but a client I'm working with has a platform where OpenAMP is allocated in cacheable memory for reasons not only related to performance improvement.

> That would means that we should have to use io access to update the virtqueues or to introduce a cmake option for the cache management.

Can you expand a bit of the cmake option? Is that an alternative to use `metal_cache_*` APIs to flush / invalidate the virtqueues when needed?

arnopo, 2021-06-08:
> I'm not sure how much I can go into details but a client I'm working with has a platform where OpenAMP is allocated in cacheable memory for reasons not only related to performance improvement.

So the rational is that they can not allocate in non-cacheable memrory, right? 
I just want to confirm with you that the need is because  your client doesn't know how to allocate non-cacheable memory. :)

> 
> > That would means that we should have to use io access to update the virtqueues or to introduce a cmake option for the cache management.
> 
> Can you expand a bit of the cmake option? Is that an alternative to use `metal_cache_*` APIs to flush / invalidate the virtqueues when needed?

Solution would consist in adding metal_cache_* but under compilation config defined depending on cmake option 
based on same mechanism as the [VIRTIO_SLAVE_ONLY](https://github.com/OpenAMP/open-amp/blob/master/cmake/options.cmake#L69) config.
The aim is to have the config disabled by default to not impact legacy but allowing to use cacheable memory by enabling it. 

For virtio buffers i suppose that it could also been implemented to have a same way of managing the cache.

carlocaione, 2021-06-08:
> So the rational is that they can not allocate in non-cacheable memrory, right?

Correct.

> I just want to confirm with you that the need is because your client doesn't know how to allocate non-cacheable memory. :)

hahah fair enough but we cannot really allocate non-cacheable memory not because we don't know how :smile: 

> Solution would consist in adding metal_cache_* but under compilation config defined depending on cmake option
based on same mechanism as the VIRTIO_SLAVE_ONLY config.
The aim is to have the config disabled by default to not impact legacy but allowing to use cacheable memory by enabling it.

Makes sense, I know that this could be a niche case after all.

PR #293 (2021-07-07): virtqueue:  Introduce cache helpers
On some platforms the VirtQueue and Vring structures could be stored in cacheable memory. When a data cache is present OpenAPM should take care of flushing / invalidating vring structures and control block when needed to keep consistency, this should be done using the metal_cache_* APIs.

This is an RFC and I'm open to any comment / suggestion on how to address this issue the proper way.

Fixes: #292
closed 2021-07-07 by arnopo