OpenAMP/open-amp #297: what compiler is used to build for this?

bentheredonethat, 2021-08-05:
https://github.com/OpenAMP/open-amp/blob/b8abc398027a324f71a408c7f4d0cba4ba243981/lib/include/openamp/rpmsg.h#L455

Per GCC manual https://gcc.gnu.org/onlinedocs/gcc-4.7.0/gcc/Function-Attributes.html 

deprecated is used as       
`          int old_fn () __attribute__ ((deprecated));`

what compiler configuration is used to allow this? or why not use the GNU GCC syntax?

Thanks
Ben

CC: @edmooring

edmooring, 2021-08-10:
For the R5 on Xilinx, GCC 8 and GCC 10 downloaded from developer.arm.com. For X86 Linux, whatever GCC came with Ubuntu LTS (16.04 and 20.04), and whatever it is Yocto uses. I'm not sure what @arnopo tested with on his systems. It appears the only environment that is having problems is whatever is in Xilinx xsct. 

The definition for __deprecated is taken from the Zephyr project and has been in use there since 2016. 

If you look at  https://gcc.gnu.org/onlinedocs/gcc-9.2.0/gcc/Attribute-Syntax.html#Attribute-Syntax you will see how this is supposed to work. 

To repeat: There is something strange going on in the way xsct uses GCC.

arnopo, 2023-05-09:
@bentheredonethat 
can we close this issue?

bentheredonethat, 2023-05-10:
@arnopo  yes. thanks