OpenAMP/open-amp #304: how to create multiple endpoints in the same channel

siliangQ, 2021-09-28:
I am working in an environment that I have an application running in Linux system and an RTOS application running on another processor. I am able to create the rpmsg channel and use the default endpoint to send messages between two applications back and forth. 
But I have some problems creating another endpoint on the same channel.
I am using **rpmsg_create_ept** to create the endpoint in the RTOS application. I think because of the name announcement service, it will not only create an endpoint, it also creates the rpmsg channel. I don't know how to disable the name announcement. And I can't find a function to just create an endpoint without creating a new channel. 
If you know how to make it work, I would really appreciate your response.

arnopo, 2021-09-29:
Hello @siliangQ 

The condition to send an ns annoucement is here [rpmsg_create_ept](https://github.com/OpenAMP/open-amp/blob/master/lib/rpmsg/rpmsg.c#L300)
```
/* Send NS announcement to remote processor */
if (ept->name[0] && rdev->support_ns &&
    ept->dest_addr == RPMSG_ADDR_ANY)
	status = rpmsg_send_ns_message(ept, RPMSG_NS_CREATE);
```
to sum-up if you don't want to use ns annoucement you can:
- not define the associated name service,
- fix the default destination endpoint address (`ept->dest_addr`) you want to communicate to.

arnopo, 2021-10-11:
Hello @siliangZ 
Does your point  answered? can i close it?

siliangZ, 2021-10-12:
Thank you so much for your reply. It works for me. Feel free to close it