OpenAMP/open-amp #315: why does open-amp need inter-core interrupts?

lenghonglin, 2021-11-15:
Hello, erveryone , We recently prepared to use OpenAmp, but now i have a big doubt.

In [RPMsg Messaging Protocol](https://github.com/OpenAMP/open-amp/wiki/RPMsg-Messaging-Protocol#physical-layer--shared-memory),  it's say `The solution proposed in this document requires only two basic hardware components - shared memory (accessible by both communicating sides) and inter-core interrupts (in a specific configuration optional).` and `In case the “used” and “avail” ring buffers have a bit set in their configuration flags field, the generation of interrupts can be completely suppressed - in such a configuration, the interrupts are not necessary. However both cores need to poll the “ring” and “used” ring buffers for new incoming messages, which may not be optimal.`

I understand that inter-core interrupts need to be used to inform the other core to receive messages, but when i read source, i find when `master` call `rpmsg_send` , then `remote` will trigger the  callback function[rpmsg_virtio_rx_callback](https://github.com/OpenAMP/open-amp/blob/3c223735ea969199b3c9e2e2a85c9fa7c022b6c0/lib/rpmsg/rpmsg_virtio.c#L469), so i can read master data from [endpoint_cb](https://github.com/zephyrproject-rtos/zephyr/blob/9ed986ae64c7679587faa8ef3608009bab8534e8/samples/subsys/ipc/openamp/src/main.c#L116). 

This means that it can run without inter-core interrupts, So what's the point of inter-core interrupts?

closed 2021-11-16 by lenghonglin

huxiangrui, 2021-12-06:
if as you said the communication is in the blocking mode, remote will check the status of the data send by server in a fixed time，it's ineffective.

arnopo, 2021-12-06:
Please create a new discussions thread instead of an issue for your questions.

The inter-core interrupts  is used to inform the remote processor that it receives a message.  
if you have not inter-core interrupt.  you can  use the polling method call rproc_virtio_notified in background to check for messages. As said @huxiangrui , not effective.