OpenAMP/open-amp #327: the doubt of the 'rpmsg_rpc_client_send() 'function

huxiangrui, 2021-12-07:
it's in the linux_rpc_demo.c.
[image]
[image]
As my understanding, those two memcpy funtion is to fill the tmpbuff in the RPMsg protocol,but the code will not run as my expected.And the tmpbuff will have the same sectors called "TERM_ID".

huxiangrui, 2021-12-07:
As my understanding, the tmpbuff should be filled in the RPMsg protocol.
[image]

arnopo, 2021-12-07:
Do you detect an issue or is a question about how to use the rpc service? In the second case please use Q&A discussions thread instead of issue.

That said, please, could you detail what do you mean by "And the tmpbuff will have the same sectors called "TERM_ID"? 
Do you try to run the demo on your Linux PC?

The linux_rpc_demo.c  uses the rpc (remote procedure call) service over the rpmsg protocol.
The tmpbuff is a concatenation of a function ID (OPEN_ID, CLOSE_ID,...) and its associated function parameters. The rpmsg_send will copies the tmpbuff in the payload of the rpmsg and fills the rpmsg header to address the message to the remote endpoint.

huxiangrui, 2021-12-14:
thanks,i got it.

arnopo, 2021-12-14:
Could we close the issue in this case ?

closed 2021-12-21 by huxiangrui