OpenAMP/open-amp #329: Elf loader, header should check for elf_state_init

thesamprice, 2022-01-03:
remote proc calls load header multiple times.
This was resulting in extra calls to loading data sections.

See
https://github.com/OpenAMP/open-amp/blob/main/lib/remoteproc/remoteproc.c#L461
https://github.com/OpenAMP/open-amp/blob/main/lib/remoteproc/remoteproc.c#L735

A possible fix is 
```
int elf_load_header(const void *img_data, size_t offset, size_t len,
                    void **img_info, int last_load_state,
                    size_t *noffset, size_t *nlen)
{
        int *load_state;

        if(last_load_state != ELF_STATE_INIT) {
                return last_load_state;
        }
```

A gist of that allows for loading elf files into memory using dev mem / linux is 
https://gist.github.com/thesamprice/aec2bda615695f1aa812ffcd3c6a8409

After fix
```bash
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff6f69700 (LWP 3762)]
metal: warning:   metal_linux_irq_handling: Failed to set scheduler: Unknown error -1.
rproc_init: node id: 1
Start to load executable with remoteproc_load()
loader_open cpu1_bram_test.elf
load_exectuable_noblock, loading 0x0,0x800
rproc_mmap: pa=0xffffffffffffffff, da=0xe8000000, size=0x28, atrribute=0x0
truncate Status is 0
Virt address f7ff4000
fread  40 bytes f7ff4000
case pa=-1: bytes_read = 40
read done 40 bytes
load_exectuable_noblock, loading 0x74,0x28
rproc_mmap: pa=0xffffffffffffffff, da=0xe8000050, size=0x111078, atrribute=0x0
Segment not alligned on a page boundary page_offset=80
truncate Status is 0
Virt address f7ec6050
fread  69656 bytes f7ec6050
case pa=-1: bytes_read = 69656
read done 69656 bytes
load_exectuable_noblock, loading 0xa0,0x11018
Load State 131586 nlen 0
Load State 131586 nlen 0
Loaded?
```
Prior fix
```
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff6f69700 (LWP 3818)]
metal: warning:   metal_linux_irq_handling: Failed to set scheduler: Unknown error -1.
rproc_init: node id: 1
Start to load executable with remoteproc_load()
loader_open cpu1_bram_test.elf
load_exectuable_noblock, loading 0x0,0x800
rproc_mmap: pa=0xffffffffffffffff, da=0xe8000000, size=0x28, atrribute=0x0
truncate Status is 0
Virt address f7ff4000
fread  40 bytes f7ff4000
case pa=-1: bytes_read = 40
read done 40 bytes
load_exectuable_noblock, loading 0x74,0x28
rproc_mmap: pa=0xffffffffffffffff, da=0xe8000050, size=0x111078, atrribute=0x0
Segment not alligned on a page boundary page_offset=80
truncate Status is 0
Virt address f7ec6050
fread  69656 bytes f7ec6050
case pa=-1: bytes_read = 69656
read done 69656 bytes
load_exectuable_noblock, loading 0xa0,0x11018
Load State 131586 nlen 0
fread  1160 bytes ffffffff
read done 1160 bytes
load_exectuable_noblock, loading 0x25ddc,0x488
Load State 131586 nlen 0
fread  292 bytes ffffffff
read done 292 bytes
load_exectuable_noblock, loading 0x25cb5,0x124
Load State 132098 nlen 0
Loaded?
```

thesamprice, 2022-01-03:
I retract that, 
I routed the metal log to printfs
```

void printf_log_handler(enum metal_log_level level,
                                  const char *format, ...){
  va_list args;
  va_start (args, format);
  vprintf (format, args);
  va_end (args);

}

...
	metal_set_log_handler(printf_log_handler);
	metal_set_log_level(METAL_LOG_DEBUG);
```

It looks like the extra load sections are the shstrtab.
```
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff6f69700 (LWP 5805)]
metal: warning:   metal_linux_irq_handling: Failed to set scheduler: Unknown error -1.
rproc_init: node id: 1
Start to load executable with remoteproc_load()
loader_open cpu1_bram_test.elf
load_exectuable_noblock, loading 0x0,0x800
remoteproc_load_noblock: check remoteproc status
remoteproc_load_noblock: check loader
Load State 0x0 nlen 0
Loading ELF headering
Loading ELF program header.
Loading ELF section header.
remoteproc_load_noblock, load header 0x0, 0x800, next 0x25ddc, 0x488
segment: 1, total segs 2
remoteproc_load_noblock, load data 0x0, 0x800, next 0x74, 0x28
rproc_mmap: pa=0xffffffffffffffff, da=0xe8000000, size=0x28, atrribute=0x0
truncate Status is 0
Virt address f7ff4000
load_exectuable_noblock ret=0x0 nlen=0x28
fread  40 bytes f7ff4000
case pa=-1: bytes_read = 40
read done 40 bytes
load_exectuable_noblock, loading 0x74,0x28
remoteproc_load_noblock: check remoteproc status
segment: 2, total segs 2
remoteproc_load_noblock, load data 0x74, 0x28, next 0xa0, 0x11018
rproc_mmap: pa=0xffffffffffffffff, da=0xe8000050, size=0x111078, atrribute=0x0
Segment not alligned on a page boundary page_offset=80
truncate Status is 0
Virt address f7ec6050
load_exectuable_noblock ret=0x0 nlen=0x11018
fread  69656 bytes f7ec6050
case pa=-1: bytes_read = 69656
read done 69656 bytes
load_exectuable_noblock, loading 0xa0,0x11018
remoteproc_load_noblock: check remoteproc status
Load State 0x20202 nlen 0
Loading ELF section header.
remoteproc_load_noblock, load header 0xa0, 0x11018, next 0x25ddc, 0x488
load_exectuable_noblock ret=0x0 nlen=0x488
fread  1160 bytes ffffffff
read done 1160 bytes
load_exectuable_noblock, loading 0x25ddc,0x488
remoteproc_load_noblock: check remoteproc status
Load State 0x20202 nlen 0
Loading ELF section header.
Loading ELF section header complete.
Loading ELF shstrtab.
remoteproc_load_noblock, load header 0x25ddc, 0x488, next 0x25cb5, 0x124
load_exectuable_noblock ret=0x0 nlen=0x124
fread  292 bytes ffffffff
read done 292 bytes
load_exectuable_noblock, loading 0x25cb5,0x124
remoteproc_load_noblock: check remoteproc status
Load State 0x20402 nlen 0
Loading ELF shstrtab.
remoteproc_load_noblock, load header 0x25cb5, 0x124, next 0x25cb5, 0x0
remoteproc_load_noblock, load data 0x25cb5, 0x124, next 0x25cb5, 0x0
Loaded?
[Thread 0x7ffff7fd9580 (LWP 5804) exited]
```

arnopo, 2022-01-04:
Hello @thesamprice 
Your point is not clear to me. do you still have an issue? if yes can you detail your current issue, else could we close it?
Thanks,
Arnaud

thesamprice, 2022-01-06:
Can close it, 

@arnopo 
Would you be opposed to including an example of a stand alone elf loader app.
A start of this is here. I could work on cleaning this up.
https://gist.github.com/thesamprice/aec2bda615695f1aa812ffcd3c6a8409

arnopo, 2022-01-07:
> Can close it,
> 
> @arnopo Would you be opposed to including an example of a stand alone elf loader app. A start of this is here. I could work on cleaning this up. https://gist.github.com/thesamprice/aec2bda615695f1aa812ffcd3c6a8409

Difficult to answer, I need more details on your objectives
Is your goal to improve the existing [exemple](https://github.com/OpenAMP/open-amp/tree/main/apps/examples/load_fw), extend it to a new platform?
If I well understood you are working on a Xilinx MicroBlaze platform, right?

thesamprice, 2022-01-07:
Yes, 
Im running on mb petalinux, mb baremetal, and planning on mb rtems.
from what i saw the load_fw were tied to the specific boards. Mostly due to startup / shutdown sequences.

One option to make it more generic would be to leave the shutdown / restart sequences out of the elf loader as those usually specific to boards.

This would allow for a generic elf loader.
The above gist was something I modified to get a generic elf loader running on mb petalinux.
I handle the shutdown / startup sequence outside of the app.

arnopo, 2022-01-07:
Yes this example seems needed to be dusted. 
Looking at https://github.com/OpenAMP/open-amp/pull/189 with hindsight, I realize that the generic part and platform part have not been not well separated.

@edmooring : Do you know why Xilinx platform specific code is directly in the example?

From my point of view. Having the generic part in open-amp/apps/examples/load_fw and the platform implementation in apps/system/linux/machine/ would be cleaner.

Seems that something has already  been implemented in this way regarding https://github.com/OpenAMP/open-amp/blob/main/apps/system/linux/machine/generic/platform_info.c. 

@thesamprice: if you have interested in improving this for Microblaze hardware you are welcome. Maybe wait for @edmooring's feedback first to get more context.

arnopo, 2023-05-09:
Hi @thesamprice,
any update on your work? should i close the issue?

thesamprice, 2023-05-09:
Close it.  I have nothing to merge back in.  I was able to run libmetal on rtems using the generic build fyi

closed 2023-05-09 by arnopo