OpenAMP/open-amp #335: remoteproc get memory function issue

joshualin-petaio, 2022-01-25:
Hi,

There has one question about get memory from remoteproc memory list.
Assume there has two memory region in the remoteproc memory list and continue.
like following description:
memory list 0 record start address (pa) : 0x00000000, size : 0x1000
memory list 1 record start address (pa) : 0x00001000, size : 0x1000
If I use remoteproc_get_io_with_pa() with pa : 0x00001000 to search memory list.
I will get memory list 0 because the get condition is match (pa + size) <= pa_end).

Does there have something wrong or need some precondition before create memory list ?

arnopo, 2022-01-25:
Hi,
Thank you for pointing this out. This need to be fixed.
could you send a PR to fix remoteproc_get_mem? to test pa < pa_end ( and same for da)?

joshualin-petaio, 2022-01-27:
Hi,

Thank you for your reply.
I had send a PR to fix remoteproc_get_mem().
(pa + size) < pa_end_end and also (da + size) < da_end.

BR,
Joshua

closed 2022-02-24 by arnopo