OpenAMP/open-amp #338: Invalidate the ring region but use non initialize ring index

joshualin-petaio, 2022-01-27:
Hi,

There have some question about use non initialized variable.
1. At the virtqueue_get_available_buffer(). In order to get latest avail index. So invalidate the head ring region. But the head index is getting after invalidate the head ring region. It may cause that wrong region will be invalidate.

2. The same issue is also happen in virtqueue_get_desc_size().

BR,
Joshua

arnopo, 2022-01-27:
I suppose that you are spekking about this lines which should be inverted?
```
	/* Avail.ring is updated by master, invalidate it */
	VRING_INVALIDATE(vq->vq_ring.avail->ring[head_idx]);

	head_idx = vq->vq_available_idx++ & (vq->vq_nentries - 1);
```
Yes that seem to me something no safe if we are at the limit of 2 cache blocks.
@carlocaione : Do you confirm the potential issue?

carlocaione, 2022-01-27:
Yes, this must be fixed.

The problem is that when we are using `VRING_INVALIDATE` the index `head_idx` is always `0`, so indeed we are invalidating the wrong ring.

Thank you for spotting this. Let me know if you are going to open a PR or whether I should do it.

joshualin-petaio, 2022-01-28:
Hi Carlocaione,

I can send a PR.
And maybe you can check later.

Thanks,
Joshua

closed 2022-02-24 by arnopo