OpenAMP/open-amp #341: Unused code in rpmsg_get_endpoint

tammyleino, 2022-02-04:
rpmsg_get_endpoint currently supports four cases for returning the endpoint:

1 - local address only
2 - local and remote address (this is actually dead code because #1 will always match first)
3 - service name and remote address
4 - service name for unregistered remote endpoint

It has been confirmed that rpmsg_get_endpoint is not an API, so I see only two potential calls to rpmsg_get_endpoint within the code, for cases 1, 3 and 4 above.

Case 1 : rpmsg_get_ept_from_addr; return rpmsg_get_endpoint(rdev, NULL, addr, RPMSG_ADDR_ANY);
Case 3 and Case 4: rpmsg_virtio_ns_callback; _ept = rpmsg_get_endpoint(rdev, name, RPMSG_ADDR_ANY, dest);

If my evaluation is correct, I propose that rpmsg_get_endpoint be modified to remove the unused functionality:

```
		/* try to find match on local end remote address */
		if (addr == ept->addr && dest_addr == ept->dest_addr)
			return ept;
```

edmooring, 2022-02-06:
If this were an official API that was intended to be externally used, I would argue that the ordering of the cases to be tested is incorrect. It should go from most restrictive to least restrictive. That would arguably be 2, 3, 4, then 1. However, since it is clearly intended to be an internal function (only declared in rpmsg_internal.h), I am fine with this proposed change.

arnopo, 2022-02-14:
Hi @tammyleino 
Could you send a Pull request to fix it?

Thanks!
Arnaud

closed 2022-04-06 by arnopo