OpenAMP/open-amp #342: rpmsg_virtio_send_offchannel_nocopy does not return error when RPMSG_ASSERT fails

tammyleino, 2022-02-04:
This routine ultimately returns the number of bytes transmitted to a calling API.

The following fail cases do not propagate a failure back up the call stack and therefore the caller will think the data has been transmitted:

```
	status = metal_io_block_write(io, metal_io_virt_to_offset(io, hdr),
				      &rp_hdr, sizeof(rp_hdr));
	RPMSG_ASSERT(status == sizeof(rp_hdr), "failed to write header\r\n");
```

. . .

```
	/* Enqueue buffer on virtqueue. */
	status = rpmsg_virtio_enqueue_buffer(rvdev, hdr, buff_len, idx);
	RPMSG_ASSERT(status == VQUEUE_SUCCESS, "failed to enqueue buffer\r\n");
```

We could either propagate the error code back up the call stack, return a generic error indicating tx failure or return 0 indicating no data was transmitted.

edmooring, 2022-02-06:
This is something worth discussing. The current approach using an assertion will kill the whole process if it fails, AND a special debug symbol is defined. Otherwise, things will attempt to progress as normal, except that an operation has failed without an error indication. I believe that the first case should be treated as a catastrophic error. The second case should return some sort of useful error code, since being out of buffers (for example) could well be a transitory condition that could succeed later.

tammyleino, 2022-02-07:
@edmooring doesn't the behavior further depend on whether RPMSG_DEBUG is defined?

arnopo, 2022-03-25:
Full agree with with @edmooring  analysis. In `rpmsg_virtio_send_offchannel_nocopy` the status returned by the `rpmsg_virtio_enqueue_buffer` function should be propagated ( transforming  VIRTQUEUE error in RPMSG error).

arnopo, 2023-05-09:
new rpmsg_release_tx_buffer has been introduced in https://github.com/OpenAMP/open-amp/pull/401 that should help to fix current issue